micropython/micropython #4616: mBoot incompatible with some application interrupt usage

andrewleech, 2019-03-19:
Using `machine.bootloader()` to directly jump to mBoot doesn't always work, depending on what interrupt based code is configured in the application.

If there are certain kinds of interrupts enabled in the application, jumping to bootloader dead-locks early in the mboot stm32_main function it tries to jump to now non-existent interrupt handlers.

We've got a number of pin external interrupts, timers running on hard interrupts and we're using the watchdog.

The bring-down code before the jump to bootloader doesn't currently include deinit for network, uart, timers or pin interrupts. 
I've tried manually disabling used ones myself and/or adding deinit C code for all of these, but have become stuck at the watchdog - it cannot be switched off without a chip reset. This leaves me at a similar point to https://github.com/micropython/micropython/pull/4413

dpgeorge, 2019-03-20:
> I've tried manually disabling used ones myself and/or adding deinit C code for all of these, but have become stuck at the watchdog - it cannot be switched off without a chip reset.

I would be happy to add deinit code in `machine.bootloader()` to shut down all peripherals, but as you say the WDT is still a problem.  And that can't be fully fixed by a system reset because it's possible to enable the IWDG in hardware via the option bits (so IWDG is started automatically upon reset).

A way to deal with this would be to have mboot feed the watchdogs.

andrewleech, 2019-03-20:
Ah, good point about the option bytes!
The initial post yesterday was half way through a long day of pulling my hair out a bit trying to nail down what the problem was, and then how to get around it. I've currently got an application level workaround orchestrating reboots similar to #4413 but done with `uctypes` in `boot.py`. Works for us for now.

I've got a set of deinit calls that's probably mostly suitable to push, though last time I tried them it was still failing. I now know the wdt was the real blocker for us however, so with that handled the rest of it is probably valid. 
I couldn't find a particularly convenient high-level method to clear all EXTI pin interrupts in a board-agnostic way, depending on how different these registers are in stmf7 to other cores. These may not have actually been an issue in the end though.

On a similar note, I've also had soft-reboot lock up trying to de-init uart while there's uart interrupts still enabled. I should also add a irq disable in the uart deinit function.

Perhaps system deinit should be abstracted to a common location and expanded a bit?
Personally I'd like it if this deinit function could be called from python, as for our board we've got a soft power off. Currently I sequence a soft-reboot to run the deinit code then detect that in boot to turn off the board. Not sure how useful a `machine.deinit()` would really be for others though.

In mboot, yes I did think feeding the watchdog is a more long term fix. 
Also, defining a stub interrupt handler that can be used for basically everything to just ignore the interrupt (and clear if needed) may help to avoid the issues with other interrupts still being enabled at times and jumping to dead space.

dpgeorge, 2019-06-25:
mboot is now entered via a full `NVIC_SystemReset()`, see 04c7cdb668cc7ee391ef5fe000f825389197f7e2

@andrewleech I think we can close this issue (although perhaps there would eventually be other things to address, in other issues, given your comment above).

andrewleech, 2019-06-27:
I can confirm your merged features works great for our application, I've been able to remove all my workarounds and bootloader is jumped to reliably.

I've been thinking more about my other issues and I think they're less important. Certainly now a full reboot is done all the peripherals are reset anyway, the main thing is that storage is flushed (already done). 

I do think it would be sensible to disable interrupts at the start of modmachine `machine_bootloader()` function to ensure that an interrupt can't jump in and affect the setup commands being used during jump to bootloader, but the rest of my concerns above don't feel as important anymore.

closed 2019-06-27 by andrewleech

dpgeorge, 2019-06-27:
> I do think it would be sensible to disable interrupts at the start of modmachine `machine_bootloader()`

Yes I agree.  That was originally done by `HAL_DeInit()` and/or `HAL_RCC_DeInit()`, which are now gone.

dpgeorge, 2019-07-04:
In afb2e9dd942a77125d4447a96def5be48fee19af IRQs were disabled before entering the bootloader.