micropython/micropython #4619: Cannot build mPy on iOS in armv7s

Ecco, 2019-03-20:
Most likely due to the same bug as #2943. I think a similar fix in `nlrthumb.c` would do the trick.

dpgeorge, 2019-03-21:
Can you please state what the error is, why it won't build?

> I think a similar fix in `nlrthumb.c` would do the trick.

I assume you mean the `jmp    _nlr_push_tail` (addition of an underscore at the start).

Ecco, 2019-03-21:
Sure. The error message is:
```
Undefined symbols for architecture armv7s:
  "nlr_push_tail", referenced from:
      _nlr_push in nlrthumb.o
     (maybe you meant: _nlr_push_tail)
```

And indeed, I meant the addition of the underscore. I have no idea what the "undo function's prelude" part of 52b6764894755324db036731ae1d6ec77172419a does 😄

dpgeorge, 2019-03-26:
Fixed by dce785cc3d79f6145dc4d4411dbef00bc6d3d9bf

closed 2019-03-26 by dpgeorge