micropython/micropython #4620: What version of ESP-IDF should I be using

johncblacker, 2019-03-20:
I want to build the ESP32 port including ESP-NOW support but don't know which version of the ESP-IDF I should have installed.  Along the same lines, which ESP xtensa compiler version should I be using?

dpgeorge, 2019-03-21:
Please see the file `ports/esp32/README.md`.  All the relevant information should be in there.

oclyke, 2019-03-23:
@johncblacker FYI the currently supported git hash of the ESP-IDF is ```5c88c5996dbde6208e3bec05abc21ff6cd822d26``` and I am using the following xtensa version (pre-built from espressif)

```
xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
```

closed 2019-03-26 by dpgeorge

johncblacker, 2019-03-27:
The problem I’ve found is that the release in the readme doesn’t always match what’s in the Makefile SUPHASH value; and, having just the SUPHASH value makes it

a chore to find the release and correct branch to use when performing the build.  I do agree, however, that if one is building the latest version, the README.md does provide

the correct information.  In my case I was trying to build from the esp-now branch, I had a SUPHASH value, but there is no way to search on github for that SUPHASH value.  

Anyway, I was able to solve my problem and complete my firmware build.  Thanks for the follow-up.

jb

 

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

dhylands, 2019-03-27:
You can just checkout the ESP-IDF using the SUPHAS value directly.

For example, this Makefile:
https://github.com/micropython/micropython/blob/92149c8a7954169285b2909012dc601c6e7cb0aa/ports/esp32/Makefile#L30
says it wants 5c88c5996dbde6208e3bec05abc21ff6cd822d26

So you can go into the ESP-IDF directory and then just do:
```
git fetch origin
git checkout 5c88c5996dbde6208e3bec05abc21ff6cd822d26
```
You don't have to care what branch or revision or whatver.

You can see which tags include the indicated commit by doing:
```
git tag --contains 5c88c5996
```
but knowing the tag or branch isn't required to build.

johncblacker, 2019-03-27:
I realize that, but one has to have the correct base release, don’t they?  If one is at the right base release and you do a git show <the suphash value> it will respond with something like:

commit 2c95a77cf93781f296883d5dbafcdc18e4389656

Merge: 2c2e7f630 552ba35da

Author: Angus Gratton <angus@espressif.com>

Date:   Tue Oct 24 10:02:51 2017 +0800

 

    Merge branch 'bugfix/ioctl_declaration' into 'master'

 

    add missing ioctl declaration

 

    See merge request !1442

 

So, then you know you can do a checkout; however, if you try that on the wrong base release you get:

fatal: not a git repository (or any of the parent directories): .git

So, obviously you cannot do a checkout.

 

It would be so much easier if there was a least the exact date of the commit…,

Thanks for the reply.

 

https://github.com/micropython/micropython/blob/92149c8a7954169285b2909012dc601c6e7cb0aa/ports/esp32/Makefile#L30

https://www.avast.com/antivirus

dhylands, 2019-03-27:
> I realize that, but one has to have the correct base release, don’t they? If one is at the right base release and you do a git show <the suphash value> it will respond with something like:

Nope. You can be on any branch. For example:
```
dhylands@dave ~/esp 542 >git clone https://github.com/espressif/esp-idf.git

Cloning into 'esp-idf'...
remote: Enumerating objects: 833, done.
remote: Counting objects: 100% (833/833), done.
remote: Compressing objects: 100% (608/608), done.
remote: Total 85802 (delta 273), reused 417 (delta 202), pack-reused 84969
Receiving objects: 100% (85802/85802), 70.73 MiB | 10.10 MiB/s, done.
Resolving deltas: 100% (61008/61008), done.
Checking out files: 100% (4372/4372), done.

dhylands@dave ~/esp 543 >cd esp-idf/

dhylands@dave ~/esp/esp-idf  (master) 544 >git show 5c88c5996dbde6208e3bec05abc21ff6cd822d26

commit 5c88c5996dbde6208e3bec05abc21ff6cd822d26
Merge: 2eabed161 6c8a8664a
Author: Jiang Jiang Jian <jack@espressif.com>
Date:   Fri Jan 25 11:53:06 2019 +0800

    Merge branch 'bugfix/ble_active_scan_not_wait_scan_rsp' into 'master'
    
    component/bt: add a option to make report adv data and scan response individually
    
    See merge request idf/esp-idf!4131

dhylands@dave ~/esp/esp-idf  (master) 545 >git checkout 5c88c5996dbde6208e3bec05abc21ff6cd822d26

Note: checking out '5c88c5996dbde6208e3bec05abc21ff6cd822d26'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at 5c88c5996 Merge branch 'bugfix/ble_active_scan_not_wait_scan_rsp' into 'master'

dhylands@dave ~/esp/esp-idf  ((5c88c5996...)) 546 >git log -1 HEAD

commit 5c88c5996dbde6208e3bec05abc21ff6cd822d26 (HEAD)
Merge: 2eabed161 6c8a8664a
Author: Jiang Jiang Jian <jack@espressif.com>
Date:   Fri Jan 25 11:53:06 2019 +0800

    Merge branch 'bugfix/ble_active_scan_not_wait_scan_rsp' into 'master'
    
    component/bt: add a option to make report adv data and scan response individually
    
    See merge request idf/esp-idf!4131
```
I added blank lines before each command to make the command stand out a bit from the results, If you already have a tree checked out and your local tree doesn't include the required commit, then you might also need to do a `git fetch origin` before doing the `git show` or `git checkout` commands.

branches and tags are really just a human readable way of specifying a commit hash.

If you get the error:
```
fatal: not a git repository (or any of the parent directories): .git
```
that means that you're not inside the git repository tree. It has nothing to do with what branch/tag/commit is currently checked out. You can switch from any branch/tag/commit to any other branch/tag/commit provided the one you're trying to switch to is in your local repository.

johncblacker, 2019-03-27:
Your last sentence seems to me to be the one that I have trouble with resolving.  I need to ensure that what I'm trying to switch to is in my local repository; therefore I need to have the correct repository cloned.  So, don't I need to know which release I have to clone?  I'm very new to git and perhaps in over my head.  If I pick the most recent repository to clone, I'm not necessarily going to have the repository that's going to have the specific commit I need; therefore, I need to know which repository to clone?

dhylands, 2019-03-27:
You don't clone releases. You clone repositories. A repository many contain many releases.

You can determine which repository you've clone by using the command:
```
git remote -v
```
When you clone a repository, it assigns that original cloned repository 'origin'.

So the repository that I cloned in the previous comment shows as:
```
dhylands@dave ~/esp/esp-idf  ((5c88c5996...)) 584 >git remote -v

origin	https://github.com/espressif/esp-idf.git (fetch)
origin	https://github.com/espressif/esp-idf.git (push)
```
I can see all of the tags (which are one way to indicate releases):
```
dhylands@dave ~/esp/esp-idf  ((5c88c5996...)) 586 >git tag -l

v0.9
v1.0
v2.0
v2.0-rc1
v2.0-rc2
v2.1
v2.1-rc1
v2.1.1
v3.0
v3.0-dev
v3.0-rc1
v3.0.1
v3.0.1-rc
v3.0.2
v3.0.3
v3.0.3-rc
v3.0.4
v3.0.4-rc1
v3.0.5
v3.0.5-rc
v3.0.6
v3.0.6-rc
v3.0.7
v3.0.7-rc
v3.1
v3.1-beta1
v3.1-dev
v3.1-rc1
v3.1-rc2
v3.1.1
v3.1.1-rc2
v3.1.2
v3.1.3
v3.2-beta1
v3.2-beta3
v3.2-dev
v3.3-beta1
v3.3-beta2
v3.3-dev
v4.0-dev
```
Once you've cloned it, your local repository will start to get stale as the maintainer makes commits to the master repository (which is perhaps on github).

You can use the `git fetch origin` command to pull down changes which have been made in the master repository which are not in your local repository.

johncblacker, 2019-03-28:
Just to close this out, I went back and downloaded the latest esp-idf github repository, setup the environment variable and path according to instructions.  I did a checkout

of the commit in the SUPHASH value in the Makefile and did a build and it worked.  I don’t know what I was doing, but somehow I got tangled up in my shorts and was making

things overly difficult for myself.  Thanks for the info about github, I needed that!  It’s not a complete mystery to me, but I’m definitely a beginner with regard to it’s usage.

I’ll try to apply what I’ve learned with this fiasco to future builds.

Thanks again.

 

John

 

 

 

https://www.avast.com/antivirus