micropython/micropython #4621: about autoindent

nbenm, 2019-03-20:
Is there a way to stop autoindent ?

dhylands, 2019-03-20:
This is a more appropriate question for the forum.

If you use paste mode (Control-E), then auto-indent will be disabled.
http://docs.micropython.org/en/latest/reference/repl.html#paste-mode

nbenm, 2019-03-20:
thanks @dhylands. In fact I'm sending instructions from a C program using uart. So copy-paste is not possible. I do this from a TI-Nspire calculator. I send commands to an esp8266-01

robert-hh, 2019-03-20:
If you can control what you send, start the transmission with Ctrl-E (which switches the device to paste mode), send the python instructions, and close it with Ctrl-D. Paste mode does not require a manual paste.

nbenm, 2019-03-20:
@robert-hh Yes I can control what I send.
I've tried your solution and it works.
Only one side effect, I have the message "paste mode; Ctrl-C to cancel,\r\n"
But I can filter it.

nbenm, 2019-03-20:
Thanks to @dhylands and @robert-hh 
Paste mode was the right answer

closed 2019-03-20 by nbenm

dhylands, 2019-03-20:
If you're doing things programmatically, then using RAW mode is probably easier. In Paste mode, the characters you send are echoed back. In raw mode, they aren't.

The pyboard.py program uses this technique. You can use pyboard.py to upload (into memory) and run python files.

[rshell](https://github.com/dhylands/rshell) uses pyboard.py to upload a small snippet of code, and then talks to that small snippet of code. So you can do things that are fairly sophisticated by using this technique.

nbenm, 2019-03-20:
I have read quickly pyboard.py.
I will have a deeper look at it. It's very interesting.
I didn't know about raw mode (I'm knew to MicroPython).
I'll try to pick some things and rewrite them in C.
I have tried sending an if...else... surrounded by Ctrl-A and Ctrl-D
I have waited for response ending with Ctrl-D twice.
As seen in pyboard.py, I receive the string "raw REPL; CTRL-B to exit\r\n" but it is followed by ">OK".
It is said in the doc that "OK" is the Ctrl-D acknowledgment.
I think in pyboard.py you read until ">" and then read again until "OK".
Everything is not completely clear for the moment. I'll take time to read your code.
But the most important is that there are nice ideas.