micropython/micropython #4626: pyboard.py remains in raw REPL when CTRL-C exit is used

rhubarbdog, 2019-03-22:
I run this simple program by typing `python pyboard.py disco.py`
File disco.py
```
import pyb

leds=[pyb.LED(i) for i in range(1,5)]

n=0
while True:
    n = ( n + 1 ) % 4
    leds[n].toggle()
    pyb.delay(50)
```
when i press CTRL-C i get my command prompt back but the program still runs on the pyboard.
I enter REPL to do some other tasks and my prompt is `>` and not `>>>`. I'm still in raw REPL

dpgeorge, 2019-03-25:
> when i press CTRL-C i get my command prompt back but the program still runs on the pyboard.

Yes, this is correct.  The ctrl-C stops just the pyboard.py script on your PC from following the output of the pyboard.  It doesn't send a ctrl-C to the pyboard itself (pyboard.py is not acting as a terminal, it just sends a file then waits for received data).

> I enter REPL to do some other tasks and my prompt is `>` and not `>>>`. I'm still in raw REPL

That is really the intended behaviour.  It might be that you want to stay in raw REPL to send some more commands.

rhubarbdog, 2019-03-25:
I thought it was confusing as the command leaves the pyboard in a different state. To novice users their serial terminal will seem broken

dpgeorge, 2019-03-25:
> To novice users their serial terminal will seem broken

The best way to address that would be with documentation about the behaviour of pyboard.py

dpgeorge, 2019-06-24:
As mentioned, this is the intended behaviour.

closed 2019-06-24 by dpgeorge