micropython/micropython #4628: pyboard.py just becomes a typewriter with no arguments or --follow

rhubarbdog, 2019-03-22:
It's unclear how to operate `pyboard.py` when executed with no arguments. I type `python pyboard.py`
I enter
```
import pyb
pyb.LED(1).on()
```
Nothing happens. I press enter a few times leaving blank lines, still nothing.  I try pressing CTRL+D which is the EOF character from the keyboard,  still nothing.

dhylands, 2019-03-22:
Pyboard.py wasn't designed to be run with no arguments. It should probably produce an error message.

dpgeorge, 2019-03-25:
> It's unclear how to operate `pyboard.py` when executed with no arguments.

The intention is that running pyboard.py with no args will make it follow the output of the attached device.  For example:
```
$ pyboard.py long_running_script.py
<watch some output, then ctrl-C to stop pyboard.py from following output>
$ do something else
$ pyboard.py
<continue watching output from device>
```
pyboard.py is not intended to be a terminal, just a way of putting a script to the device, then watching the output of that script running.

closed 2019-06-24 by dpgeorge