micropython/micropython #4631: while compiling micropython , A fatal error: invalid segment count 25 (max 16)

kenjican, 2019-03-23:
Detailed error message as below:
GEN build/esp32.common.ld
LINK build/application.elf
text data bss dec hex filename
843420 214080 27868 1085368 108fb8 build/application.elf
Create build/application.bin
esptool.py v2.6

A fatal error occurred: Invalid segment count 25 (max 16). Usually this indicates a linker script problem.
Makefile:844: recipe for target 'build/application.bin' failed
make: *** [build/application.bin] Error 2

my environment:
OS: Ubuntu 18.04 64-bits
xtensa-esp32-elf-gcc version : gcc version 5.2.0 (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a)
esp-idf commit hash :commit 5c88c5996dbde6208e3bec05abc21ff6cd822d26 ,the same as micropython's makefile
micropython : master branch

What should I do to make compiling move on? Thanks~~

dpgeorge, 2019-03-25:
According to https://www.esp32.com/viewtopic.php?f=13&p=39999 it might help to run `git submodule update --init`

kenjican, 2019-03-25:
Thanks for the advice. I've git submodule update --init , then make clean, make.  Error remains the same

dpgeorge, 2019-03-25:
The only thing I can suggest is to try and build an ESP IDF example app (one with wifi enabled) to see if that works for your OS and tool chain.

kenjican, 2019-03-26:
Thanks man! , I execute git submodule update (without --init) again ..and make again. it works~~

closed 2019-03-26 by kenjican