micropython/micropython #4633: UART reading not working in ESP8266 with v1.10

sysseon, 2019-03-24:
After a lot of testing, I found that UART reading with ESP8266 doesn't work at all in version 1.10. It doesn't raise any exception, but the UART cannot read any character. After downgrading to v1.9.4, it all worked.

I am using UART0 (the only one in the ESP8266 that has RX). Writing works, but reading doesn't (`uart.any()` returns 0, `uart.read()` returns `None`...). I've checked this problem (same code and conditions) with two ESP8266 (Wemos D1 mini) and MicroPython v1.9.3, v1.9.4 and v1.10 (including today release: 20190324). The problem is only present in the latter.

I have checked also with a logic analyzer that the attached device is sending the information correctly to the ESP8266's RX pin (`GPIO3`). However, in v1.10 I cannot sniff the RX pin, so maybe the problem is related with the pinout configuration.

Maybe I am missing something that has recently changed in the UART library, because I think this is a pretty serious bug for a stable version...

dpgeorge, 2019-03-25:
> Maybe I am missing something that has recently changed in the UART library,

Yes, in v1.10 the UART was changed so it could more easily be used as a standalone UART, not attached to the REPL.  Please see #3784 for more info.  To get it working may require `uos.dupterm(None, 1)`

rhubarbdog, 2019-03-25:
I have a similar issue on the pyboard v1.1. 
I believe `uart.any()` works, but `uart.read()` (read all) does nothing until it tries to allocate 80+Kb and crashes.
solution is : 
```
amount = uart.any()
if amount > 0:
    data = uart.read(amount)
```

sysseon, 2019-03-25:
Thank you Damien, I missed that change. Now reading is working fine, and I can receive a 0x03 byte without the program terminating :) For my part, this issue can be closed.

dpgeorge, 2019-03-26:
> I have a similar issue on the pyboard v1.1.
> I believe `uart.any()` works, but `uart.read()` (read all) does nothing until it tries to allocate 80+Kb and crashes.

@rhubarbdog Probably you just need to set the timeout to 0: `UART(1, 9600, timeout=0)`.  Then uart.read() will return once there's a gap in the incoming stream of chars.

closed 2019-03-26 by dpgeorge