micropython/micropython #4634: Further ways to improve .mpy encoding

pfalcon, 2019-03-24:
Following up after https://github.com/micropython/micropython/pull/4564 .

* https://github.com/micropython/micropython/issues/4378 remains an issue.
* After #4564, the situation is such that there're 2 sufficiently differing (== independent) encodings for the bytecode: one is in-memory encoding, as used by VM, another is .mpy. .mpy is noticeably transformed during loading process to get to in-memory encoding. But that only means that further transformations can be added. One such would be not storing extra cache bytes for name lookup opcodes in .mpy - indeed, it's useless there, and needed only by VM. That means that MICROPY_OPT_CACHE_MAP_LOOKUP_IN_BYTECODE flag can be gone from .mpy header, and .mpy's can be made a bit more portable (fully portable between "beefy" ports of MicroPython which use unicode, MPZ long ints, as long as small ints are limited to 32 bits). But that shifts responsibility of handling MICROPY_OPT_CACHE_MAP_LOOKUP_IN_BYTECODE to other tools, like mpy-tool.py.

dpgeorge, 2019-04-08:
> One such would be not storing extra cache bytes for name lookup opcodes in .mpy - indeed, it's useless there, and needed only by VM.

Yes that should be possible, to just insert them into the bytecode as it's loaded.

> That means that MICROPY_OPT_CACHE_MAP_LOOKUP_IN_BYTECODE flag can be gone from .mpy header

That would be great thing to do, but I don't think it's possible.  The point being that bytecode jump offsets will be different depending on whether they were compiled/calculated with or without cache bytes.  So to do it would require the bytecode loader to recompute these jump offsets, which is complicated and/or memory consuming.

Although, bytecode could always be encoded assuming no caching, and so only ports that do have caching would need the complexity of rewriting the jump offsets.  Given the benefits of much more portable bytecode, it might be worth the complexity to rewrite offsets.

dpgeorge, 2022-06-23:
The .mpy format was significantly reworked in f2040bfc7ee033e48acef9f289790f3b4e6b74e5.  Bytecode is now mostly static data.  The MICROPY_OPT_CACHE_MAP_LOOKUP_IN_BYTECODE flag is also gone.

closed 2022-06-23 by dpgeorge