micropython/micropython #4639: Does STM32F4DISC support SDCART ?

ahmdrz, 2019-03-25:
Hi,
Does STM32F407VG (discovery) support SDCart? Is it a feature that will be added in the future? If not, can I do anything to make this happen? If so, how?

ahmdrz, 2019-03-25:
Can I use it alone? Not in discovery board. Only STM32F407VG microcontroller.

dpgeorge, 2019-03-26:
The STM32F4DISC board doesn't have an SD card slot.  But the software does support SD card on the STM32F407 if you enable it with MICROPY_HW_HAS_SDCARD.  See boards/PYBV10/mpconfigboard.h for an example.

ahmdrz, 2019-03-26:
So I think https://www.kaltpost.de/?p=2199 will help me with this issue. Right?

dpgeorge, 2019-03-26:
Yes that post should help.

For these kinds of questions (that are not bugs) it's best to ask at https://forum.micropython.org

closed 2019-03-26 by dpgeorge