micropython/micropython #4640: Building with MICROPY_PY_WIZNET5K=1 fails

diginfo, 2019-03-26:
```
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
GEN build-PYBV11/genhdr/pins.h
GEN stmconst build-PYBV11/modstm_qstr.h
make: *** No rule to make target 'drivers/wiznet5k/ethernet/w1/w1.c', needed by 'build-PYBV11/genhdr/qstr.i.last'.  Stop.
```

I have checked the drivers/wiznet5k/ethernet folder and there is no such subfolder w1 ??

What am I missing ??

dpgeorge, 2019-03-26:
In e36821a7662e9c65ee2bfb8c23925a10d2167faf it was changed so you need to specify the chip number, eg `MICROPY_PY_WIZNET5K=5500`

diginfo, 2019-03-26:
Great, Thanks Damien, but now fails with:

```
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
mkdir -p build-PYBV11/drivers/wiznet5k/ethernet/
mkdir -p build-PYBV11/drivers/wiznet5k/ethernet/w5500/
mkdir -p build-PYBV11/drivers/wiznet5k/internet/dns/
CC network_wiznet5k.c
CC modnwwiznet5k.c
In file included from ../../py/objlist.h:29:0,
                 from modnwwiznet5k.c:31:
modnwwiznet5k.c:476:19: error: 'MP_QSTR_ifconfig' undeclared here (not in a function)
     { MP_ROM_QSTR(MP_QSTR_ifconfig), MP_ROM_PTR(&wiznet5k_ifconfig_obj) },
                   ^
../../py/obj.h:92:56: note: in definition of macro 'MP_OBJ_NEW_QSTR'
 #define MP_OBJ_NEW_QSTR(qst) ((mp_obj_t)((((mp_uint_t)(qst)) << 2) | 2))
                                                        ^
modnwwiznet5k.c:476:7: note: in expansion of macro 'MP_ROM_QSTR'
     { MP_ROM_QSTR(MP_QSTR_ifconfig), MP_ROM_PTR(&wiznet5k_ifconfig_obj) },
       ^
modnwwiznet5k.c:485:17: error: 'MP_QSTR_WIZNET5K' undeclared here (not in a function)
         .name = MP_QSTR_WIZNET5K,
                 ^
modnwwiznet5k.c:485:9: error: initializer element is not constant
         .name = MP_QSTR_WIZNET5K,
         ^
modnwwiznet5k.c:485:9: error: (near initialization for 'mod_network_nic_type_wiznet5k.base.name')
../../py/mkrules.mk:47: recipe for target 'build-PYBV11/modnwwiznet5k.o' failed
make: *** [build-PYBV11/modnwwiznet5k.o] Error 1
```

dpgeorge, 2019-03-26:
It needs a `make clean`.

diginfo, 2019-03-26:
OOps, yes that did the trick, thanks so much

closed 2019-03-26 by diginfo

diginfo, 2019-03-26:
Sorrie, to ask here, but are you able to offer any solution to this:

https://forum.micropython.org/viewtopic.php?f=6&t=6176

The adapter is working but not recognized by any modules that import socket ?!