micropython/micropython #4645: umqtt not complied into firmware.bin

kenjican, 2019-03-27:
All other py file in modules folder are complied into firmware.bin, except umqtt.  I noticed that umqtt is a folder , simple.py and robust.py are in it. Which part of Makefile  shall I modify to make umqtt compiled ?

peterhinch, 2019-03-27:
This is a query best raised in [the forum](http://forum.micropython.org/index.php). I assume you're using ESP8266. The umqtt library is organised as a Python package. Packages should compile correctly by default. You should be able to issue
```python
from umqtt import simple
```
It works here on ESP8266 running release build V1.10.

kenjican, 2019-03-27:
Thanks Peter,,It's ESP32 and I just found why. umqtt is linked to micropython-lib but I didn't git clone it. Since I git clone micropython-lib and make again...umqtt comes out~

closed 2019-03-27 by kenjican