micropython/micropython #4647: esp32 onewire not working since 2019-03-25 updates

kevinkk525, 2019-03-27:
I am using the daily build downloads for the esp32 (without psram) and the 2019-03-25 is working fine with ds18x20 sensors but the new build 2019-03-27 always gives me CRC errors. 
I can only guess what might be causing this but as within the last few days only one small change to threading and the activation of dual core usage was changed, the problem should lie within these 2 changes.

How to reprodue:
- Connect DS18x20
- Try to read using firmware 2019-03-25
- Confirm that reading works without problems
- Flash firmware 2019-03-27
- Confirm that no reading works, always CRC error

dpgeorge, 2019-03-29:
Thanks for the report, but I can't reproduce the issue.  Using the latest commit 0fb15fc3f4b532faa3ca4fe49809f9d9e6c5cd53 (which includes dual core) I can read 1 or 2 sensors reliably on a LoLin32 Pro.

Remember that the DS18x20 needs 5v to power it, and a pull-up resistor on the data line (to 5v).  In my test I didn't use an external pull-up, the esp32's internal one was enough (and is enabled by default by the OneWire driver).

kevinkk525, 2019-04-01:
Thanks for testing it.
This is however very strange. The version 2019-03-27 does definitely not work for me. I get at most 1 ds18b20 to send its temperature correctly and none of my custom onewire slaves. It's also very unstable.

With version 2019-03-25 everything worked fine and with the new version 2019-04-01 it does too although I don't see a new commit for the 04-01 version..

I tried this multiple times with the same setup and the same test files. Just flashed another firmware in between.

Therefore as the new version works correctly again, there's probably no reason to speculate about the 03-27 not working for me and I'll close this issue.

dpgeorge, 2019-04-01:
This probably deserves more investigation.  The performance of the esp32 (or certain functions running on it) can vary by a factor of 2 depending on where things get located in ROM, due to caching.  So one build may behave differently to another, just because of how it's compiled.

Fixing it might require putting the onewire code in iRAM.