micropython/micropython #4651: enabling MICROPY_PY_LWIP breaks MICROPY_PY_WIZNET5K

diginfo, 2019-03-28:
Just confirmed this:

**MICROPY_PY_LWIP + MICROPY_PY_WIZNET5K**
```
import network, socket
net = network.WIZNET5K(pyb.SPI(1), pyb.Pin('X5'), pyb.Pin('X4'))
net.active(True)
net.ifconfig(('192.168.0.18','255.255.255.0','192.168.0.1','8.8.8.8'))
while not net.isconnected():
	pass
net.ifconfig(('192.168.0.18','255.255.255.0','192.168.0.1','8.8.8.8'))
net.ifconfig()
socket.getaddrinfo('micropython.org',80)
```
**Fails:**
```
('192.168.0.18', '255.255.255.0', '192.168.0.1', '8.8.8.8')
Traceback (most recent call last):
  File "<stdin>", line 9, in <module>
OSError: -2
```

**MICROPY_PY_WIZNET5K only - success**

```
>>> socket.getaddrinfo('micropython.org',80)
[(2, 1, 0, '', ('176.58.119.26', 80))]
```

So it looks like MICROPY_PY_LWIP somehow is breaking MICROPY_PY_WIZNET5K

In both cases ifconfig returns the correct IP, however when using LWIP the device cannot be pinged and is unable to access the network.

diginfo, 2019-03-28:
I can run the simple webserver example and it does not error during startup, but are unable to connect to it.

Also, I have found that I can ping the device, but that's all.

`socket.getaddrinfo('micropython.org',80)`

Fails every time, not sure what other info I can provide ?

dpgeorge, 2019-03-29:
The lwIP mode works for me, using a PYBv1.0+Wiz550io.  The following is enough for it to work, using DHCP by default:
```python
import network, socket
lan = network.WIZNET5K(pyb.SPI(1), pyb.Pin.board.X5, pyb.Pin.board.X4)
lan.active(1)
while not lan.isconnected():
    pass
print(lan.ifconfig())
print(socket.getaddrinfo('micropython.org', 80))
```
It also seems to work fine for me using a static IP address.

> Also, I have found that I can ping the device, but that's all.

If you can ping it then the physical connection is there, it has an IP address, and the lwIP stack is working.  So it should then be able to do everything else.

Make sure you use the latest, unmodified code from this repo.

pacmac, 2019-04-05:
Hi Damien;

I am still having problems with the wiznet module on the PYBOARD11.

The current problem is that the wiznet module retains it's first DHCP configuration and does not appear to get a new lease when connecting to a completely different network.

Is there a way to force it to renew it's DHCP lease ?

Thanks

dpgeorge, 2019-04-05:
> Is there a way to force it to renew it's DHCP lease ?

Try `lan.ifconfig('dhcp')`

pacmac, 2019-04-05:
Thanks I am getting a :

`OSError: timeout waiting for DHCP to get IP address`

```
import socket as s
s.getaddrinfo('micropython.org',80)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: -2
```
This combined with the previous issue where it is unable to resolve dns leads me to believe that there may some problem with UDP packets ??

In order to eliminate hardware issue, I ordered and just received a bunch of new WIZNET devices, however I am getting the same results on the old and new hardware ?

pacmac, 2019-04-05:
I can connect to the device from a browser using a simple web server, but it looks like outbound connections have a problem ?!

dpgeorge, 2019-04-05:
> In order to eliminate hardware issue, I ordered and just received a bunch of new WIZNET devices, however I am getting the same results on the old and new hardware ?

What are the model numbers of these, is it a 5500 chip?

pacmac, 2019-04-05:
Yes, both old and new are 5500 - would you like to login ? - I can set that up

dpgeorge, 2019-04-05:
Please contact me by email (see git log for it).

pacmac, 2019-04-05:
I am having problems connecting the PYBOARD to the ubuntu machine for you to login to:

```
dmesg
[   86.174022] usb 2-2-port1: attempt power cycle
[   88.154102] usb 2-2.1: new full-speed USB device number 10 using uhci_hcd
[   88.180832] usb 2-2.1: device descriptor read/8, error 2
[   88.313327] usb 2-2.1: device descriptor read/all, error 8
```

Have you seen this issue before ? - it is a fresh install of UBU18

dpgeorge, 2019-04-05:
> Have you seen this issue before ? - it is a fresh install of UBU18

The USB port could be overloaded (power wise).  Or the USB Linux driver has gotten into a strange state and no longer works, and the machine must be rebooted (this happens to me sometimes, especially with many repeated pyboard plug/unplug cycles).

pacmac, 2019-04-05:
Solved it by using a USB hub, just refused to enumerate otherwise.

Sent you an email with the login details

dpgeorge, 2019-04-05:
The issue was that the Wiznet device didn't have a MAC address assigned.  This case is now handled by fd523c53c37b36a9c343b1061d08e65e920dffa0; see also 4f936afc441da48ab8043ec108a2a2f6a92fdd6e

closed 2019-04-05 by dpgeorge

pacmac, 2019-04-05:
Probably all of these $5.00 China WIZ devices don't have mac addresses assigned.

LukeB1675, 2019-06-21:
Hello all,

I have just started looking into micropython for development. I recently began working with the WIZ820io off a PYBV1.1. I am having all the exact same problems as Diginfo had plus the attempt below. If I build firmware using only the MICROPY_PY_WIZNET5K=5200 option all seems to work fine. Except "socket" doesn't allow for the method "makefile" used in the server example in the docs. I am running out of ideas to get around this. Any help would be greatly appreciated!

>>> import network, socket
>>> lan = network.WIZNET5K(pyb.SPI(1), pyb.Pin.board.X5, pyb.Pin.board.X4)
>>> lan.active(1)
>>> wiznet5k_lwip_poll: fatal error len=482 ret=-1001
wiznet5k_send_ethernet: fatal error -7
wiznet5k_send_ethernet: fatal error -7
wiznet5k_send_ethernet: fatal error -7
wiznet5k_send_ethernet: fatal error -7
wiznet5k_send_ethernet: fatal error -7

-Thank you in advance, Luke.

dpgeorge, 2019-06-24:
> I recently began working with the WIZ820io

The 5500 module has better support than the 5200, so you might want to try that instead.

eslamdiab400, 2021-02-25:
i need to connect enc28j60 with esp32 but i dont find WIZNET5K in network.
what should i do??

dpgeorge, 2021-03-11:
> i need to connect enc28j60 with esp32 but i dont find WIZNET5K in network.
> what should i do??

Wiznet is only supported on the stm32 port at the moment, not esp32.  On esp32 you can use `network.LAN` instead.

eslamdiab400, 2021-03-11:
Ok thank you