micropython/micropython #4654: How to add pip modules to micropython?

Edris89, 2019-03-29:
Hi there,

I was wondering if it's possible to add installs/modules from using pip.
A example would be the pip install catt module. 

So my question is, is it possible to add modules, compile and make the bin file to flash it on the esp8266. With the extra module added.

If so could somebody point me to the right direction?

peterhinch, 2019-03-29:
This question should be raised in [the forum](http://forum.micropython.org/index.php) not in this MicroPython bug tracker. You might first like to read [the documentation](http://docs.micropython.org/en/latest/reference/packages.html).

jonnor, 2023-07-30:
MicroPython has a package manager called `mpip`. Usage is described here: https://docs.micropython.org/en/latest/reference/packages.html
Proposing to close this issue.

closed 2023-08-03 by jimmo