micropython/micropython #4655: esp32 compile failed

ray-x, 2019-03-29:
Greetings
I am trying to compile ESP32 firmware.  However using master branch of micropython failed the build.
The ESP-IDF version is :
commit 5c88c5996dbde6208e3bec05abc21ff6cd822d26 (HEAD -> micropython)
And bellow is error log. It seems that pyparsing not installed.
$ make SDKCONFIG=boards/sdkconfig 
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
GEN build/esp32.common.ld
Traceback (most recent call last):
  File "/home/ray/Projects/esp/esp-idf/tools/ldgen/ldgen.py", line 22, in <module>
    from fragments import FragmentFileModel
  File "/home/ray/Projects/esp/esp-idf/tools/ldgen/fragments.py", line 20, in <module>
    from sdkconfig import SDKConfig
  File "/home/ray/Projects/esp/esp-idf/tools/ldgen/sdkconfig.py", line 18, in <module>
    from pyparsing import Word, printables, Combine, Literal, hexnums, quotedString, Optional, nums, removeQuotes, oneOf, Group, infixNotation, opAssoc
ModuleNotFoundError: No module named 'pyparsing'

But pyparsing install correctly. When run the last command (line 18 of sdkconfig:
>>> from pyparsing import Word, printables, Combine, Literal, hexnums, quotedString, Optional, nums, removeQuotes, oneOf, Group, infixNotation, opAssoc
>>> dir(Word)
['DEFAULT_WHITE_CHARS', '_FifoCache', '_UnboundedCache', '__add__', '__and__', '__call__', '__class__', '__delattr__', '__dict__', '__doc__', '__eq__', '_.....

Did anyone see this error before?

BTW, ESP-IDF can compile correctly with hello_world in examples folder.

Thanks!
Ray

dpgeorge, 2019-03-29:
You may need to explicitly set the Python version, like `make PYTHON=python2`

ray-x, 2019-03-29:
Thanks George for your feed back
with "make PYTHON=python2"  the compile completed correctly.
I think this should be documented in Readme.md for ESP32 port.
Currently I am using python virtual environment built from esp-idf/requirement.txt and the python version is python2.7.15.  Not sure why PYTHON=python2 need to be specified during build.
My linux is Ubuntu18.10.

Thanks!
Ray

dpgeorge, 2019-03-29:
> with "make PYTHON=python2" the compile completed correctly.
> I think this should be documented in Readme.md for ESP32 port.

This is something that all the ports may require (depending on how your build system is set up) and is mentioned in the main README.md in the root of this repository.

ray-x, 2019-03-29:
I saw the main README.md did mentioned PYTHON=python2
But it is strange that I built port/unix before and did not specify PYTHON=python2 does not cause similar issue. I can not verify STM32. But the behaviors are different for ESP32 and unix.
I could close the ticket but I suspect it will surface again for those following the instructions in port/ESP32.

dpgeorge, 2019-04-01:
> But it is strange that I built port/unix before and did not specify PYTHON=python2 did not cause similar issue.

Yes that is strange.  So this could be addressed in the esp32's README, probably in the paragraph about Python 2 vs 3 and installing pyserial (sounds like a similar issue).

supersjimmie, 2019-04-08:
I am confused about this.
The main README.md says:
```
"make" is used to build the components, or "gmake" on BSD-based systems. You will also need bash, gcc, and Python 3.3+ available as the command python3 (if your system only has Python 2.7 then invoke make with the additional option PYTHON=python2).
```
The way that I read this, I only need to specify ```PYTHON=python2``` if I don't have python3 (only 2.7).
My system has both 2.7.12 and 3.5.2 but I still need to add the python2 part.

Is there any other consequence on using python2 instread of 3 (except for avoiding this error)?

dpgeorge, 2019-04-09:
I think what's required here is that all the Python packages needed for the ESP32 must be installed for the version of Python that is used.

Since Python 3 is the default, the following packages must be installed *for Python 3*: pyserial, pyparsing.

If you explicitly specify Python 2, via `make PYTHON=python2`, then pyserial and pyparsing must be installed *for Python 2*.

Other ports don't require these packages so that's why they build fine with Python 2 or Python 3 without any trouble.

A paragraph in the esp32's README would be enough to clarify this point.

supersjimmie, 2019-04-09:
Thanks, yes pyparsing was missing here for python3.

dpgeorge, 2019-04-18:
> A paragraph in the esp32's README would be enough to clarify this point.

Done in 4ce0091449052daca592f852a31eece074d34a57

closed 2019-04-18 by dpgeorge