micropython/micropython #4656: pyboard-D EMMC Not Visible

diginfo, 2019-03-29:
I have the 4GB EMMC module connected to my pyboard D, however I am unable to see the 4GB flash storage:

```
cd /Volumes/PYBFLASH
df -h
Filesystem      Size   Used  Avail Capacity iused      ifree %iused  Mounted on
/dev/disk1s1   2.0Mi  100Ki  1.9Mi     5%     512          0  100%   /Volumes/PYBFLASH
```

Am I missing something ??

diginfo, 2019-03-29:
OK, figured this out which does not error, but I still can't see the flash ?!

```
import os, pyb, sys
os.mount(pyb.MMCard(), '/mm')
sys.path.append('/mm')
```

diginfo, 2019-03-29:
The format script appears to contain some errors:

This is not a bootable disk, please insert a bootable file...

```
mmc.mmc_format()
Formatting...
000001b0 00 00 00 00 00 00 00 00 98 96 af 9c 00 00 00 21 |...............!|
000001B0 00 00 00 00 00 00 00 00 98 96 AF 9C 00 00 00 21
000001c0 03 00 06 17 ea fd 00 08 00 00 00 f8 71 00 00 00 |............q...|
000001C0 03 00 06 17 EA FD 00 08 00 00 00 F8 71 00 00 00
000001f0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 55 aa |..............U.|
000001F0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 55 AA
0
00100000 eb 58 90 6d 6b 66 73 2e 66 61 74 00 02 08 20 00 |.X.mkfs.fat... .|
00100000 EB 58 90 6D 6B 66 73 2E 66 61 74 00 02 08 20 00
00100010 02 00 00 00 00 f8 00 00 3e 00 76 00 00 08 00 00 |........>.v.....|
00100010 02 00 00 00 00 F8 00 00 3E 00 76 00 00 08 00 00
00100020 00 f8 71 00 70 1c 00 00 00 00 00 00 02 00 00 00 |..q.p...........|
00100020 00 F8 71 00 70 1C 00 00 00 00 00 00 02 00 00 00
00100030 01 00 06 00 00 00 00 00 00 00 00 00 00 00 00 00 |................|
00100030 01 00 06 00 00 00 00 00 00 00 00 00 00 00 00 00
00100040 80 00 29 88 8c 78 e6 4e 4f 20 4e 41 4d 45 20 20 |..)..x.NO NAME |
00100040 80 00 29 88 8C 78 E6 4E 4F 20 4E 41 4D 45 20 20
00100050 20 20 46 41 54 33 32 20 20 20 0e 1f be 77 7c ac | FAT32  ...w|.|
00100050 20 20 46 41 54 33 32 20 20 20 0E 1F BE 77 7C AC
00100060 22 c0 74 0b 56 b4 0e bb 07 00 cd 10 5e eb f0 32 |".t.V.......^..2|
00100060 22 C0 74 0B 56 B4 0E BB 07 00 CD 10 5E EB F0 32
00100070 e4 cd 16 cd 19 eb fe 54 68 69 73 20 69 73 20 6e |.......This is n|
00100070 E4 CD 16 CD 19 EB FE 54 68 69 73 20 69 73 20 6E
00100080 6f 74 20 61 20 62 6f 6f 74 61 62 6c 65 20 64 69 |ot a bootable di|
00100080 6F 74 20 61 20 62 6F 6F 74 61 62 6C 65 20 64 69
00100090 73 6b 2e 20 20 50 6c 65 61 73 65 20 69 6e 73 65 |sk. Please inse|
00100090 73 6B 2E 20 20 50 6C 65 61 73 65 20 69 6E 73 65
001000a0 72 74 20 61 20 62 6f 6f 74 61 62 6c 65 20 66 6c |rt a bootable fl|
001000A0 72 74 20 61 20 62 6F 6F 74 61 62 6C 65 20 66 6C
001000b0 6f 70 70 79 20 61 6e 64 0d 0a 70 72 65 73 73 20 |oppy and..press |
001000B0 6F 70 70 79 20 61 6E 64 0D 0A 70 72 65 73 73 20
001000c0 61 6e 79 20 6b 65 79 20 74 6f 20 74 72 79 20 61 |any key to try a|
001000C0 61 6E 79 20 6B 65 79 20 74 6F 20 74 72 79 20 61
001000d0 67 61 69 6e 20 2e 2e 2e 20 0d 0a 00 00 00 00 00 |gain ... .......|
001000D0 67 61 69 6E 20 2E 2E 2E 20 0D 0A 00 00 00 00 00
001001f0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 55 aa |..............U.|
001001F0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 55 AA
100000
```

diginfo, 2019-03-29:
Ahhh..

```
>>> pyb.MMCard().info()
(3825205248, 512, 0)
>>> 

>>> pyb.MMCard().present()
False
```
Looks like it can't see the EMMC !

diginfo, 2019-03-29:
This is **NOT** urgent, I am just reporting what I found.

dpgeorge, 2019-04-01:
> os.mount(pyb.MMCard(), '/mmc')
> sys.path.append('/mmc')
> pyb.usb_mode('VCP+MSC', msc=(pyb.MMCard(),))

This (above) is all you should need in boot.py to get the MMC working (it doesn't need an explicit power(1) call).  It can't be auto-detected like the SD card so needs manual configuration.

Note that once the USB is set up you can't reconfigure it without first disabling it via `pyb.usb_mode(None)`.  So best to put this in your boot.py.

diginfo, 2019-04-01:
Thanks, yes done, that several times, it does not appear to do anything:

```
>>> import os, pyb
>>> os.mount(pyb.MMCard(), '/mmc')
>>> import sys
>>> sys.path.append('/mmc')
>>> pyb.usb_mode('VCP+MSC', msc=(pyb.MMCard(),))
>>> os.listdir()
['main.py', 'lib', 'm.py', '.fseventsd', 'b.py', '.Spotlight-V100', 'setup.py', 'old', 'boot.py', 'board.py', 'apps', 'data', 'html', 'board_pyb_d.py', '.Trashes']
>>> 
```

dpgeorge, 2019-04-01:
After mounting the eMMC it won't automatically chdir to it, you'll need to do `os.chdir('/mmc')`, or just use absolute paths like `os.listdir('/mmc')`

pacmac, 2019-04-01:
Hi;

This is my boot.py file:

```
import os, pyb, sys
os.mount(pyb.MMCard(), '/mmc')
sys.path.append('/mmc')
pyb.usb_mode('VCP+MSC', msc=(pyb.MMCard(),))
```

From the REPL I can see an empty mount:

```
os.chdir('/mmc')
os.listdir()
```

However I cannot see or read/write to this mount from my macbook:

```
cd /Volumes/PYBFLASH
ls -all
drwxrwxrwx@ 1 pac   staff  16384  1 кві 20:25 .
drwxr-xr-x@ 5 root  wheel    160  1 кві 20:17 ..
drwxrwxrwx  1 pac   staff   1024  1 кві 20:25 .TemporaryItems
-rwxrwxrwx  1 pac   staff   4096  1 кві 20:25 ._boot.py
-rwxrwxrwx@ 1 pac   staff    120  1 кві 20:25 boot.py
```

So how do I access the mount from my macbook / imac etc ?

Also what is the procedure to reset the filesystem on the PYBOARD-D ?

pacmac, 2019-04-01:
OK, solved.

If I hold down the USR button when doing a quick-press RESET, the EMMC is mounted and I can read/write to it.

Thanks for your patience :-)

dpgeorge, 2019-04-02:
> If I hold down the USR button when doing a quick-press RESET, the EMMC is mounted and I can read/write to it.

That shouldn't be necessary every time.  Once boot.py is saved and unmounted, a normal RST (or plugging in the board) should let the eMMC be visible to the PC.

> Is it possible to name the volume as it mounts as unnamed, and maybe default the name to PYBMMC ?

I've now changed it to PYBMMC in the mmc_format.py script.  You should be able to change it yourself from the PC.

closed 2019-04-02 by dpgeorge