micropython/micropython #4662: [PYBD] Switching Frequency completly lovks up the board

ThomasChr, 2019-04-01:
Examplecode:
```machine.freq(216000000)```
More details in a Forum Thread: https://forum.micropython.org/viewtopic.php?f=6&t=6190

dpgeorge, 2019-04-01:
I replied on the forum, and I can't reproduce the problem, even switching at a USB REPL works fine.

robert-hh, 2019-04-01:
At the first attempt it usually works. If you switch between two frequencies several times at REPL, it will block after a few attempts.

ThomasChr, 2019-04-01:
I'll try to explain it as thoroughly as I can, because I know everything can be important.
1. I hook up my Pyboard D (SF2W) via USB. The cable I used works fine with every other Board.
    - Neither boot.py nor main.py aren't changed since I got the board fresh. main.py is completly empty (apart from a comment), boot.py has importet machine and pyb and set the country to 'US' (which is wrong on germany but won't be a problem for this test, I guess)
2. I see a new drive in the Windows Explorer "PYBFLASH". (I can access it just fine)
3. I have a new COM Port in the Device Manager (COM38).
4. I fire up putty (0.70) which has worked with all other boards without problems.
5. The connection Parameters I use are: Serial, COM 38, 115200 Baud, Flow control: None
6. Pushing 'open' opens a blank black window. I seemed that the Pyboard has resetted somehow because it didn't greet me with an empty window, instead is printed the startup message and the prompt:
```
MicroPython v1.10-418-g8ef236ec5-dirty on 2019-03-26; PYBD_SF2W with STM32F722IEK
Type "help()" for more information.
>>>
```
I enter
```
machine.freq(216000000)
```
and push enter.
The cursor goes to the next line and there is no new prompt:
```
MicroPython v1.10-418-g8ef236ec5-dirty on 2019-03-26; PYBD_SF2W with STM32F722IEK
Type "help()" for more information.
>>> machine.freq(216000000)

```
This is reliable every time I tried. It didn't work once!

I now press enter a few times: No change- also the cursor doesn't move.
I now close the putty window and try to reconnect with the same parameters: I get a black window with the cursor in the upper left corner. Nothing more. No prompt. Nothing. Pushing enter does not help.
I tried to access the PYBFLASH Drive. I can refresh it, save files with text and read them back. (Of course we can't be really sure if that's not caching of the OS, but I doubt it, it seems to react just fine).

After pushing the reset button the Flash drive vanishes and reapears and I can connect again with putty.

I haven't tried it on linux, so I can't know how it will behave there.

If there is anything I can try to help you figuring this out I will, just let me know!

Thomas

robert-hh, 2019-04-01:
Putty on Windows has the problem, that it does not recover when there was a problem with the USB_VCP interface.

ThomasChr, 2019-04-01:
That maybe the case, but even restarting putty and trying to reconnect does not work.

robert-hh, 2019-04-01:
So I tried it again with picocom on Linux, alternating between two frequencies. After the 4th or 5th time in a row (slowly entered via REPL) it locks up.

ThomasChr, 2019-04-01:
I‘m very glad to see that this bug is not only out of my imagination! Thanks Robert!

robert-hh, 2019-04-01:
Ok. Here is a little test code. What happens. after a few print statements, the output stops, but the LED continues to toggle. If I terminate picocom and restart it immediately, the output starts over for a few lines. As a related effect, sometimes the counter i is not printed.
```
import time, machine
from pyb import LED
led = LED(1) # 1=red, 2=green, 3=blue
for i in range(20):
    machine.freq(192000000)
    led.toggle()
    print(i, machine.freq())
    time.sleep(1)
    machine.freq(120000000)
    led.toggle()
    print(i, machine.freq())
    time.sleep(1)
```

robert-hh, 2019-04-01:
The same script runs on OS X with picocom without any problem. So it's more a problem of the USB driver of the PC.

dpgeorge, 2019-04-02:
Ok, thanks for all the details.  It sounds like different PC's have different behaviour with the USB.  What could be improved here is to make a "smoother" transition between frequencies for the USB peripheral, eg synchronise the frequency switch just after the USB start-of-frame, and disable/pause the USB during the switch.

Otherwise, in the meantime, set the frequency once in boot.py, or change it only with USB disabled/unused (eg `pyb.usb_mode(None)`).

ThomasChr, 2019-04-02:
Setting the frequency in boot.py helps on Windows but I need to use HTerm. Putty does not work... strange...

Thanks for the tip!

pacmac, 2019-04-29:
is this related to https://github.com/micropython/micropython/commit/8031b7a25c21fb864fe9dd1fa40740030be66c11 ??

dpgeorge, 2019-04-30:
> is this related to 8031b7a25c21fb864fe9dd1fa40740030be66c11 ?

It could be.  That commit may help improve the situation with USB and frequency selection, but it's still possible that USB is lost when changing frequency because it misses a request from the PC during the switch.