micropython/micropython #4664: [ESP8266] Timer period is limited to 7 mins with v1.10

JanBednarik, 2019-04-01:
After flashing v1.10 on ESP8266 (board Wemos D1 mini v3.1.0) timers are limited to cca 7 mins.

I haven't checked exact limit, but when set to 7m and 10s, the callback won't be executed.

```python
t1 = Timer(-1)
t1.init(period=1000 * 60 * 7, callback=lambda t: print("07:00"))

t2 = Timer(-1)
t2.init(period=1000 * 60 * 7 + 1000 * 10, callback=lambda t: print("07:10"))
```

With MicroPython v1.9.4 on the same board periods like 20+ mins works fine (I did not check what's the limit).

dpgeorge, 2019-05-07:
Thanks for the report, I can confirm the issue.  It looks like the regression was introduced by c3c914f4dded3224d27ee73ce044ac43de11c2fb with the addition of the call to `system_timer_reinit()`, that this function changes the resolution of the timer to microseconds, and then limits the maximum period to 428496 milliseconds.

dpgeorge, 2019-05-07:
Apart from reverting that commit and removing the ability to have sub-millisecond timer precision, the only way to fix this would be to synthesise periods above the maximum.

dpgeorge, 2019-05-10:
Fixed by 99a8fa794089de2ca7292a49bdcb8a4550a06f4c

closed 2019-05-10 by dpgeorge