micropython/micropython #4665: pyboard-D Board Definition and Firmware Download

pacmac, 2019-04-01:
I have just flashed my pyboard-D with pyboard 1.1 firmware as I wrongly (stupidly) assumed they were the same / universal as there is no alternate board definition in the stm32/builds folder.

Can you provide me with a link to the STM32F767 firmware download as the board is now unresponsive and obviously needs flashing with the correct firmware.

Also, is there a board definition for the pyboard-d yet so that I can use my frozen bytecode and build my own ?

Many Thanks

ThomasChr, 2019-04-01:
Same as: https://github.com/micropython/micropython/issues/4663

dpgeorge, 2019-04-02:
Closing as duplicate of #4663

closed 2019-04-02 by dpgeorge

dpgeorge, 2019-04-02:
> Can you provide me with a link to the STM32F767 firmware download as the board is now unresponsive and obviously needs flashing with the correct firmware.

Please find it here: https://micropython.org/download#pybd