micropython/micropython #4666: Unix port doesn't compile: ‘nlr_push’ undeclared here

mbirth, 2019-04-02:
The compilation of the Unix port always aborts with this error:

```
../py/nativeglue.c:226:5: error: ‘nlr_push’ undeclared here (not in a function); did you mean ‘nlr_jump’?
     nlr_push,
     ^~~~~~~~
     nlr_jump
make[1]: *** [../py/mkrules.mk:47: build/py/nativeglue.o] Error 1
make[1]: Leaving directory '/opt/micropython/mpy-cross'
make: *** [../../py/mkrules.mk:108: ../../mpy-cross/mpy-cross] Error 2
```

Tried with v1.9.4 and with a fresh master checkout.

ThomasChr, 2019-04-02:
No problems on Ubuntu 18.04.1 LTS and Raspberry Pi 2B with Rasbian 8:
```
apt-get install libffi-dev build-essential pkg-config
mkdir micropython
cd micropython
git clone https://github.com/micropython/micropython .
git submodule update --init
cd ports
cd unix
make
./micropython
```

robert-hh, 2019-04-02:
@mbirth Did you run after cloning the command
`git submodule update --init`
in the micropython directory?

mbirth, 2019-04-02:
@robert-hh Of course!

Here's the log: [micropython-compile-log.txt](https://github.com/micropython/micropython/files/3033964/micropython-compile-log.txt)

```
libffi-dev:
  Installed: 3.2.1-8
```

dpgeorge, 2019-04-04:
It sounds like the compiler is not auto-detecting the NLR variant to use, and falling back to MICROPY_NLR_SETJMP.

This line here should be evaluating to true in the C preprocessor: https://github.com/micropython/micropython/blob/master/py/nlr.h#L55

dpgeorge, 2019-04-04:
> I'm running aarch64.

The native emitter doesn't support 64-bit ARM.  Two options to get it building are:
- disable MICROPY_EMIT_ARM (I'm not even sure why this is enabled if it's aarch64)
- compile in 32-bit mode via `make MICROPY_FORCE_32BIT=1`

mbirth, 2019-06-02:
I just compiled it successfully for ARM64 on Ubuntu 19.04.

closed 2019-06-02 by mbirth