micropython/micropython #4667: Print traceback without an exception

mzakharocsc, 2019-04-02:
Hello,  in CPython, there is a way to print a traceback outside of an exception, through `traceback.print_stack()` function. What would it take to add similar functionality to micropython?

dhylands, 2019-04-02:
Probably the simplest way would be to just raise an exception inside a try block.
```
import sys

class MyException(Exception):
    pass

def foo1():
    try:
        raise MyException('traceback')
    except MyException as err:
        sys.print_exception(err)

def foo2():
    foo1()

foo2()
```
I get this result:
```
>>> import print_exc2
Traceback (most recent call last):
  File "print_exc2.py", line 8, in foo1
MyException: traceback
```

mzakharocsc, 2019-04-02:
unfortunately, this method only has traceback up to foo1, it does not include foo2, which is what I am after -> full traceback from the start of the interpreter execution to the function calling print_stack()

dhylands, 2019-04-03:
I suspect that may just be due to some optimization (and foo1 being really simple). Normally the tracebacks I see go all the way up.

mzakharocsc, 2019-04-03:
There is no optimization involved. Only uncaught exceptions go all the way up.

dhylands, 2019-04-03:
Right. It only prints the traceback from the try block until the raise. So I had to modify the code to look like this:
```
import sys

class MyException(Exception):
    pass

def foo1():
    raise MyException('traceback')

def foo2():
    foo1()

def foo3():
    foo2()

try:
    foo3()
except BaseException as err:
    print('Caught exception at top')
    sys.print_exception(err)
```
and then I got this output:
```
>>> import print_exc2
Caught exception at top
Traceback (most recent call last):
  File "print_exc2.py", line 16, in <module>
  File "print_exc2.py", line 13, in foo3
  File "print_exc2.py", line 10, in foo2
  File "print_exc2.py", line 7, in foo1
MyException: traceback
>>> 
```
IIRC this is because the call stack is collected as the stack is unwound.

mzakharocsc, 2019-04-03:
still does not work if in foo2() you have a generic
 ```
try:
    foo1()`
except:
    pass
```
The traceback would not be displayed at all in this case. I need a stack trace regardless of whether code in upper functions catches and silences exceptions or not.

Does anyone know where in micropython codebase, the variable, relating to the stack is, and how to traverse it?

dpgeorge, 2019-04-04:
To get the string of a traceback, given the exception object, you can use:
```python
buf = io.StringIO()
sys.print_exception(exc, buf)
print(buf.getvalue())
```

> I need a stack trace regardless of whether code in upper functions catches and silences exceptions or not.

Can you provide an example of how you would do this in CPython?

mzakharocsc, 2019-04-04:
`sys.print_exception()` can either be silenced by tri/catch, or wont go far enough to print the whole stack (depending on usage). Also, having to traverse the exception stack can have unintended consequences, for example, executing code inside `finally:` blocks, which may be undesirable. Sometimes you just want to print the stack right there and then and not execute anything else.    `traceback.print_stack()` in CPython accomplishes just that.  Below is a CPython example, illustrating `print_stack()` usage, and along it, the limitations of trying to do this using `sys.print_exception()`
```
import traceback
import io

def foo1():
    buf = io.StringIO()
    traceback.print_stack(file=buf)
    print("the good stuff")
    print(buf.getvalue())
    try:
        raise Exception("Not full traceback")
    except BaseException as exc:
        buf = io.StringIO()
        traceback.print_exc(file=buf)
        print("too short")
        print(buf.getvalue())
    raise Exception("Silenced traceback")

def foo2():
    try:
        foo1()
    except:
        pass

try:
    foo2()
except BaseException as exc:
    buf = io.StringIO()
    sys.print_exception(exc, buf)
    print(buf.getvalue())
else:
    print("no traceback")
```

and I get the following output in CPython 3.5:
```
the good stuff
  File "foo.py", line 25, in <module>
    foo2()
  File "foo.py", line 20, in foo2
    foo1()
  File "foo.py", line 6, in foo1
    traceback.print_stack(file=buf)

too short
Traceback (most recent call last):
  File "foo.py", line 10, in foo1
    raise Exception("Not full traceback")
Exception: Not full traceback

no traceback
```

mzakharocsc, 2019-04-04:
I have a first-pass of implementing `print_stack()`: mzakharocsc/micropython@1b847b0900e12ac69c472b4546d3c94acbafc837

I have to admit, I have no idea what I am doing, it seems to work on a few examples I threw at it. Not sure whether this works in all cases. I would like to know if I can avoid creating a global `g_current_state` variable to store a linked-list of function call stack. Is this information accessible from somewhere else? Comments/concerns please welcome.

dpgeorge, 2019-05-14:
MicroPython and CPython handle tracebacks differently:
- In MicroPython a raised exception is annotated with traceback information as it propagates up the call chain (as @dhylands mentions above).  There is no concept of a traceback object separate from an exception, they are one and the same thing.
- In CPython exception objects and traceback objects are independent: an exception is raised and as it propagates up a traceback object is created.  CPython then allows you to capture/create a full traceback object at any point in the code.

@mzakharocsc as you found in your implementation of `print_stack()` it's a bit tricky to try and separate these concepts (exception and traceback), since the VM contains the code which extracts line number info.  So, to implement this feature properly, would first need a fair bit of refactoring of how tracebacks are created, potentially changing to the way CPython handles these things.