micropython/micropython #4670: SF6 - Build Fails at last hurdle

pacmac, 2019-04-03:
Just tried to build, I thought it was going to succeed, but:

```
....
CC ../../lib/mbedtls/library/x509_csr.c
CC ../../lib/mbedtls/library/x509write_crt.c
CC ../../lib/mbedtls/library/x509write_csr.c
CC ../../lib/mbedtls/library/xtea.c
CC build-PYBD_SF6/pins_PYBD_SF6.c
LINK build-PYBD_SF6/firmware.elf
arm-none-eabi-ld: error: build-PYBD_SF6/firmware.elf uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/4.9.3/libgcc.a(bpabi.o) does not
arm-none-eabi-ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/4.9.3/libgcc.a(bpabi.o)
arm-none-eabi-ld: error: build-PYBD_SF6/firmware.elf uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/4.9.3/libgcc.a(_divdi3.o) does not
arm-none-eabi-ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/4.9.3/libgcc.a(_divdi3.o)
arm-none-eabi-ld: error: build-PYBD_SF6/firmware.elf uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/4.9.3/libgcc.a(_udivdi3.o) does not
arm-none-eabi-ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/4.9.3/libgcc.a(_udivdi3.o)
Makefile:584: recipe for target 'build-PYBD_SF6/firmware.elf' failed
make: *** [build-PYBD_SF6/firmware.elf] Error 1
```

ThomasChr, 2019-04-03:
Had no problem building!
(did not try flashing to the SF6 though...)
```
sudo apt-get install build-essential gcc-arm-none-eabi binutils
mkdir micropython
cd micropython
git clone https://github.com/dpgeorge/micropython .
git checkout stm32-pybd
git submodule update --init
cd ports/stm32
make BOARD=PYBD_SF6
```

ThomasChr, 2019-04-03:
Here the outcome:
```
LINK build-PYBD_SF6/firmware.elf
   text    data     bss     dec     hex filename
 914552     148   93328 1008028   f619c build-PYBD_SF6/firmware.elf
Create build-PYBD_SF6/firmware.dfu
GEN build-PYBD_SF6/firmware.hex
```
Building on Ubuntu 18.04

pacmac, 2019-04-03:
I am using the same build environment as for all other build for the past 1 year, and all other ports build successfully ?!

pacmac, 2019-04-03:
I justv ran this and it failed with identical error:

```
git clone https://github.com/dpgeorge/micropython micropython-pybd
cd micropython-pybd
git checkout stm32-pybd
git submodule update --init
cd ports/stm32
make BOARD=PYBD_SF6
```

```
uname -a
Linux ubuntu 4.4.0-142-generic #168-Ubuntu SMP Wed Jan 16 21:00:45 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
```

ThomasChr, 2019-04-03:
Here my version:
```
/micropython/ports/stm32$ arm-none-eabi-gcc -v
Using built-in specs.
COLLECT_GCC=arm-none-eabi-gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-none-eabi/6.3.1/lto-wrapper
Target: arm-none-eabi
Configured with: ../src/configure --build=x86_64-linux-gnu --prefix=/usr --includedir='/usr/lib/include' --mandir='/usr/lib/share/man' --infodir='/usr/lib/share/info' --sysconfdir=/etc --localstatedir=/var --disable-silent-rules --libdir='/usr/lib/lib/x86_64-linux-gnu' --libexecdir='/usr/lib/lib/x86_64-linux-gnu' --disable-maintainer-mode --disable-dependency-tracking --mandir=/usr/share/man --enable-languages=c,c++,lto --enable-multilib --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-shared --disable-threads --disable-tls --build=x86_64-linux-gnu --target=arm-none-eabi --with-system-zlib --with-gnu-as --with-gnu-ld --with-pkgversion=15:6.3.1+svn253039-1build1 --without-included-gettext --prefix=/usr/lib --infodir=/usr/share/doc/gcc-arm-none-eabi/info --htmldir=/usr/share/doc/gcc-arm-none-eabi/html --pdfdir=/usr/share/doc/gcc-arm-none-eabi/pdf --bindir=/usr/bin --libexecdir=/usr/lib --libdir=/usr/lib --disable-libstdc++-v3 --host=x86_64-linux-gnu --with-headers=no --without-newlib --with-multilib-list=rmprofile CFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-iopiMw/gcc-arm-none-eabi-6.3.1+svn253039=. -fstack-protector-strong' CPPFLAGS='-Wdate-time -D_FORTIFY_SOURCE=2' CXXFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-iopiMw/gcc-arm-none-eabi-6.3.1+svn253039=. -fstack-protector-strong' FCFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-iopiMw/gcc-arm-none-eabi-6.3.1+svn253039=. -fstack-protector-strong' FFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-iopiMw/gcc-arm-none-eabi-6.3.1+svn253039=. -fstack-protector-strong' GCJFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-iopiMw/gcc-arm-none-eabi-6.3.1+svn253039=. -fstack-protector-strong' LDFLAGS='-Wl,-Bsymbolic-functions -Wl,-z,relro' OBJCFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-iopiMw/gcc-arm-none-eabi-6.3.1+svn253039=. -fstack-protector-strong' OBJCXXFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-iopiMw/gcc-arm-none-eabi-6.3.1+svn253039=. -fstack-protector-strong' INHIBIT_LIBC_CFLAGS=-DUSE_TM_CLONE_REGISTRY=0 AR_FOR_TARGET=arm-none-eabi-ar AS_FOR_TARGET=arm-none-eabi-as LD_FOR_TARGET=arm-none-eabi-ld NM_FOR_TARGET=arm-none-eabi-nm OBJDUMP_FOR_TARGET=arm-none-eabi-objdump RANLIB_FOR_TARGET=arm-none-eabi-ranlib READELF_FOR_TARGET=arm-none-eabi-readelf STRIP_FOR_TARGET=arm-none-eabi-strip
Thread model: single
gcc version 6.3.1 20170620 (15:6.3.1+svn253039-1build1)
```

pacmac, 2019-04-03:
@ThomasChr - I guess you are on ubuntu 18, I am on 16.04 LTS - but I can build other ports without a problem.

pacmac, 2019-04-03:
It appears (from google) to related to floating point problems, I have tried adding:

`-mfloat-abi=hard -mfpu=vfp`

to CFLAGS but it makes no difference.

ThomasChr, 2019-04-03:
Getting the same error on Ubuntu 16.04 with following gcc version:
```
arm-none-eabi-gcc -v
Using built-in specs.
COLLECT_GCC=arm-none-eabi-gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-none-eabi/4.9.3/lto-wrapper
Target: arm-none-eabi
Configured with: ../src/configure --build=x86_64-linux-gnu --prefix=/usr --includedir='/usr/lib/include' --mandir='/usr/lib/share/man' --infodir='/usr/lib/share/info' --sysconfdir=/etc --localstatedir=/var --disable-silent-rules --libdir='/usr/lib/lib/x86_64-linux-gnu' --libexecdir='/usr/lib/lib/x86_64-linux-gnu' --disable-maintainer-mode --disable-dependency-tracking --mandir=/usr/share/man --enable-languages=c,c++,lto --enable-multilib --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-shared --disable-threads --disable-tls --build=x86_64-linux-gnu --target=arm-none-eabi --with-system-zlib --with-gnu-as --with-gnu-ld --with-pkgversion=15:4.9.3+svn231177-1 --without-included-gettext --prefix=/usr/lib --infodir=/usr/share/doc/gcc-arm-none-eabi/info --htmldir=/usr/share/doc/gcc-arm-none-eabi/html --pdfdir=/usr/share/doc/gcc-arm-none-eabi/pdf --bindir=/usr/bin --libexecdir=/usr/lib --libdir=/usr/lib --disable-libstdc++-v3 --host=x86_64-linux-gnu --with-headers=no --without-newlib --with-multilib-list=armv6-m,armv7-m,armv7e-m,armv7-r CFLAGS='-g -O2 -fstack-protector-strong' CPPFLAGS=-D_FORTIFY_SOURCE=2 CXXFLAGS='-g -O2 -fstack-protector-strong' FCFLAGS='-g -O2 -fstack-protector-strong' FFLAGS='-g -O2 -fstack-protector-strong' GCJFLAGS='-g -O2 -fstack-protector-strong' LDFLAGS='-Wl,-Bsymbolic-functions -Wl,-z,relro' OBJCFLAGS='-g -O2 -fstack-protector-strong' OBJCXXFLAGS='-g -O2 -fstack-protector-strong' INHIBIT_LIBC_CFLAGS=-DUSE_TM_CLONE_REGISTRY=0 AR_FOR_TARGET=arm-none-eabi-ar AS_FOR_TARGET=arm-none-eabi-as LD_FOR_TARGET=arm-none-eabi-ld NM_FOR_TARGET=arm-none-eabi-nm OBJDUMP_FOR_TARGET=arm-none-eabi-objdump RANLIB_FOR_TARGET=arm-none-eabi-ranlib READELF_FOR_TARGET=arm-none-eabi-readelf STRIP_FOR_TARGET=arm-none-eabi-strip
Thread model: single
gcc version 4.9.3 20150529 (prerelease) (15:4.9.3+svn231177-1)
```
it works fine on a new Ubuntu 18.04!

pacmac, 2019-04-03:
Thanks, Thomas that's what i suspected

dhylands, 2019-04-03:
@pacmac it looks like you have a really old toolchain. I'd recommend that you upgrade using one from here: https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm

pacmac, 2019-04-03:
Dave, thatks but that is the latest toolchain available on ubuntu 16.04LTS

pacmac, 2019-04-03:
I just updated to 6.5, however it is still using 4.93 to compile:

add-apt-repository ppa:ubuntu-toolchain-r/test
apt update
apt install gcc-6

```
LINK build-PYBD_SF6/firmware.elf
arm-none-eabi-ld: error: build-PYBD_SF6/firmware.elf uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/4.9.3/libgcc.a(bpabi.o) does not
arm-none-eabi-ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/4.9.3/libgcc.a(bpabi.o)
```

pacmac, 2019-04-03:
Solution on 16.04:

```
apt remove binutils-arm-none-eabi gcc-arm-none-eabi libnewlib-arm-none-eabi

wget https://developer.arm.com/-/media/Files/downloads/gnu-rm/8-2018q4/gcc-arm-none-eabi-8-2018-q4-major-linux.tar.bz2?revision=d830f9dd-cd4f-406d-8672-cca9210dd220?product=GNU%20Arm%20Embedded%20Toolchain,64-bit,,Linux,8-2018-q4-major

tar -xvf gcc-arm-none-eabi-8-2018-q4-major-linux.tar.bz2?revision=d830f9dd-cd4f-406d-8672-cca9210dd220?product=GNU%20Arm%20Embedded%20Toolchain,64-bit,,Linux,8-2018-q4-major

export PATH=$PATH:/root/gcc-arm-none-eabi-8-2018-q4-major/bin/
```

dhylands, 2019-04-03:
You could have also done your PATH like:
```
export PATH=/root/gcc-arm-none-eabi-8-2018-q4-major/bin/:$PATH
```
and then you don't have to uninstall the system toolchain

pacmac, 2019-04-03:
Flash is successful, but reports 1.9.3 which I assume is correct for damien's repo:

```
MicroPython v1.9.3-1708-g4417c03-dirty on 2019-04-04; PYBD_SF6W with STM32F767IIK
Type "help()" for more information.
>>>
```

dpgeorge, 2019-04-04:
I can confirm the issue on Ubuntu 16.04.  It looks like the following line in stm32/Makefile doesn't find the correct libgcc (probably due to an old gcc not understanding the FPU flags):
```
LIBS = $(shell $(CC) $(CFLAGS) -print-libgcc-file-name)
```
Try forcing the correct version using the following:
```
make BOARD=PYBD_SF2 LIBS=/usr/lib/gcc/arm-none-eabi/4.9.3/armv7e-m/fpu/libgcc.a
```

pacmac, 2019-04-04:
Just tried disabling the previous eabi and adding the libs to the make:

```
extint.c:(.text.Handle_EXTI_Irq+0xa0): undefined reference to `cyw43_poll'
build-PYBD_SF2/cyw43/cyw43_lwip.o: In function `cyw43_tcpip_link_status':
cyw43_lwip.c:(.text.cyw43_tcpip_link_status+0x14): undefined reference to `cyw43_wifi_link_status'
Makefile:584: recipe for target 'build-PYBD_SF2/firmware.elf' failed
make: *** [build-PYBD_SF2/firmware.elf] Error 1
```

dpgeorge, 2019-04-04:
Try this (for SF2 model, should also work with SF6):
```
make BOARD=PYBD_SF2 LIBS="/usr/lib/gcc/arm-none-eabi/4.9.3/armv7e-m/fpu/libgcc.a cyw43/libcyw43.a"
```

pacmac, 2019-04-04:
Yes, that works, will that also work for pyb1.1 so I can have a common GNUMakefile ?

pacmac, 2019-04-04:
No, it fails, so my script will need to include this for SFX only.

dpgeorge, 2019-04-04:
> will that also work for pyb1.1 so I can have a common GNUMakefile ?

I don't think so, due to the libcyw43.a in the LIBS.  But at the moment you can't build PYBv1.1 with the PYBD branch, that needs to be fixed first.

pacmac, 2019-04-04:
is it possible to have a conditional if ... in the GNUMakefile to test whether the board is an SF Board ?

pacmac, 2019-04-04:
I am not building on the PYBD branch, I have scripts that automate linking of GNUMakefile as well as frozen files based on the port-board that is supplied to the script as a arg

dpgeorge, 2019-04-04:
> is it possible to have a conditional if ... in the GNUMakefile to test whether the board is an SF Board ?

```
ifeq ($(BOARD),PYBD_SF6)
# stuff for SF6
endif
```

pacmac, 2019-04-04:
Thanks, I will close this now.

closed 2019-04-04 by pacmac