micropython/micropython #4672: time module Import Error

ARM93, 2019-04-04:
Hi, i have successfully compile and run the micropython source code from "https://os.mbed.com/users/infinnovation/code/micropython/#33521d742af1". but when import the time module then i faced below error.

>>> import sys                                                                         
>>> import time                                                                        
Traceback (most recent call last):                                                     
  File "<stdin>", in <module>                                                          
ImportError: module not found                                                          
>>> while :1
... print('hello')
... time.sleep_ms(2000)
... 
Traceback (most recent call last):
  File "<stdin>"
SyntaxError: invalid syntax
>>> 

kindly help to resolve this issue.

robert-hh, 2019-04-04:
What happens if you try `import utime as time`?

ARM93, 2019-04-04:
thanks for your response. i got below result when tried to import the utime.
>>> import utime
Traceback (most recent call last):
  File "<stdin>", in <module>
ImportError: module not found
>>>

dhylands, 2019-04-04:
It appears that port of micropython has yet to implement the time module.

dhylands, 2019-04-04:
It also appears that the author has abandoned the port as there hasn't been any changes in 3 years.

ARM93, 2019-04-04:
> It also appears that the author has abandoned the port as there hasn't been any changes in 3 years.

Hi  Dhylands, Even the while also not worked.So i believe the standard c functionalities are need to implemented. 

Could you please guide me to port that repository further to implement standard c functionalities.

dpgeorge, 2019-04-08:
> Even the while also not worked.

You have a typo there, swapped ":" and "1", it should be: `while 1:`

> Could you please guide me to port that repository further to implement standard c functionalities.

You can try https://github.com/micropython/micropython-infineon

dpgeorge, 2019-04-08:
Sorry, that last link to micropython-infineon is not correct.  To get the source for that mbed port you can try following the .tar.gz links on the mbed users page.

I'll close this issue because it's not related to the repository here.

closed 2019-04-08 by dpgeorge