micropython/micropython #4673: WDT spams error in when using the latest firmware esp32-20190404-v1.10-261-gd89ce2ed1.bin

vindolin, 2019-04-04:
WDT throws an error every five seconds when using the latest firmware **esp32-20190404-v1.10-261-gd89ce2ed1.bin** 

```python
import time

time.sleep(5)  # don't lock ourselves out

print('starting!')

i = 0

while True:
    if not i % 100:
        print(int(i / 100))
    i += 1
    time.sleep(0.01)
```

```
starting!
0
1
2
3
E (20148) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (20148) task_wdt:  - IDLE1 (CPU 1)
E (20148) task_wdt: Tasks currently running:
E (20148) task_wdt: CPU 0: IDLE0
 CPU 1: mp_taskwdt
4
6
7
8
E (25148) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (25148) task_wdt:  - IDLE1 (CPU 1)
E (25148) task_wdt: Tasks currently running:
E (25148) task_wdt: CPU 0: IDLE0
E (25148) task_wdt: CPU 1: mp_task
9
10
11
12 ...
```

I tried calling machine.idle() and wdt.feed() but that didn't help.

Rolling back to **esp32-20190125-v1.10.bin** fixed the problem.

dpgeorge, 2019-04-04:
Thanks for the report.  I think it's needed to add the following to the esp32/boards/sdkconfig file:
```
CONFIG_TASK_WDT_CHECK_IDLE_TASK_CPU1=n
```

robert-hh, 2019-04-04:
That seems to fix it. I had that with another test script, which was running w/o keyboard interaction.

dpgeorge, 2019-04-05:
Thanks for confirming the fix.  Fixed in 69cb24a21d1df509b90f8c32b2224b84dde36f33

closed 2019-04-05 by dpgeorge