micropython/micropython #4674: esp32 make keeps failing on frozen_mpy

supersjimmie, 2019-04-06:
Running make on ports/esp32.
I tried to add some different dirs as ```FROZEN_MPY_DIR``` to the make command, but it keeps failing:

```
GEN build/frozen_mpy.c
Traceback (most recent call last):
  File "../../tools/mpy-tool.py", line 865, in <module>
    main()
  File "../../tools/mpy-tool.py", line 853, in main
    raw_codes = [read_mpy(file) for file in args.files]
  File "../../tools/mpy-tool.py", line 721, in read_mpy
    return read_raw_code(f, qstr_win)
  File "../../tools/mpy-tool.py", line 703, in read_raw_code
    return RawCodeBytecode(fun_data.buf, qstrs, objs, raw_codes)
  File "../../tools/mpy-tool.py", line 426, in __init__
    super().__init__(MP_CODE_BYTECODE, bytecode, 0, qstrs, objs, raw_codes)
TypeError: super() takes at least 1 argument (0 given)
../../py/mkrules.mk:122: recipe for target 'build/frozen_mpy.c' failed
make: *** [build/frozen_mpy.c] Error 1
make: *** Deleting file 'build/frozen_mpy.c'
```

supersjimmie, 2019-04-06:
Can be closed, I made a mistake when I tried to fix another issue (python2 var) that caused this new issue,

supersjimmie, 2019-04-06:
Could this be caused by the commit that changed that code:
https://github.com/micropython/micropython/commit/ea3c80a514c5dc4cc3a8349815eceec4fa1ac57f#diff-646c0b0930e677b56657448197cd79d0

dpgeorge, 2019-04-07:
Thanks for the report.  I think the problem is that the recent changes to mpy-tool.py means it stopped working with Python 2.

dpgeorge, 2019-04-08:
Should be fixed by 643d2a0e86bd151130508171804e5ab5b1a7bd4e

closed 2019-04-08 by dpgeorge