micropython/micropython #4675: Invalid micropython decorator on Unix port

COOLIRON2311, 2019-04-06:
My program with micropython.asm_thumb always crashes on line 4 with error message:

SyntaxError: invalid micropython decorator

I’m using Unix port on Raspberry Pi 3B.
So my questions are the following:
1). Is there even an ability to use it on Unix port?
2). If yes, does it depend on architecture? Can I use it on Pi?
3). If both yes, and there is some additional actions needed during building of micropython, what and how needs to be done?

dpgeorge, 2019-04-08:
> My program with micropython.asm_thumb always crashes on line 4 (definition of function) with error message:

The reason for that is because inline assembler is not enabled in the unix port (for any arch).

> Is there even an ability to use it on Unix port?

Yes.

> If yes, does it depend on architecture? Can I use it on Pi?

Yes, but only Thumb and Xtensa inline assembler are supported.  Probably you can use Thumb on a Pi.

>  If both yes, and there are some additional actions needed during building of micropython, what and how needs to be done?

Yes there are additional config options that need to be set.  Try this for the unix port:
```
$ make -DMICROPY_EMIT_INLINE_THUMB=1
```

COOLIRON2311, 2019-04-08:
Yes, according to raspberry pi forums Thumb2 works on Pi 3.

> ```
> $ make -DMICROPY_EMIT_INLINE_THUMB=1
> ```

Fails with:
```
make: invalid option — 'D'
make: invalid option — 'M'
```

```
make -e DMICROPY_EMIT_INLINE_THUMB=1 micropython -j4
```
Successfully builds micropython, but the same error occurred when runnning program with inline assembler.

Would someone kindly give me proper make command to build micropython?

dpgeorge, 2019-04-08:
Sorry, it should be:
```
$ make CFLAGS_EXTRA='-DMICROPY_EMIT_INLINE_THUMB=1'
```
Or you can add `#define MICROPY_EMIT_INLINE_THUMB (1)` manually to ports/unix/mpconfigport.h

COOLIRON2311, 2019-04-08:
Thank you! The build succeeded, but now other errors occur.

The example program:
```
#!/usr/bin/env micropython

@micropython.asm_thumb
def f():
    movw(r0, 42)

print(f())
```

Exits with ```Illegal instruction```

And the other one:
```
#!/usr/bin/env micropython

@micropython.asm_thumb
def f(r0, r1):
    add(r0, r0, r1)

print(f(1, 2))
```

Exits with ```Bus error```

Any ideas about the reasons of it? (if I am not spamming too much)

dpgeorge, 2019-04-08:
So here it gets a bit more complicated.  You first need to determine if `MICROPY_EMIT_THUMB` or `MICROPY_EMIT_ARM` is being auto-configured in `ports/unix/mpconfigport.h`.  Then if it's the latter you need to also define:
```
#define MICROPY_MAKE_POINTER_CALLABLE(p) ((void*)((mp_uint_t)(p) | 1))
```
This should get asm_thumb decorators working.  But it will break native/viper decorators.

Also, the executable should probably be 32-bit (that's probably already the case).

COOLIRON2311, 2019-04-08:
- The resulting executable is 32-bit ARM ELF

- There are:
```
#if !defined(MICROPY_EMIT_THUMB) && defined(__thumb2__)
    #define MICROPY_EMIT_THUMB      (1)
    #define MICROPY_MAKE_POINTER_CALLABLE(p) ((void*)((mp_uint_t)(p) | 1))
#endif
// Some compilers define __thumb2__ and __arm__ at the same time, let
// autodetected thumb2 emitter have priority.
#if !defined(MICROPY_EMIT_ARM) && defined(__arm__) && !defined(__thumb2__)
    #define MICROPY_EMIT_ARM        (1)
```

In `ports/unix/mpconfigport.h`. 

- Are they being auto configured while I'm running `make`?
Or should I manually configure them and how?

dpgeorge, 2019-04-09:
> Are they being auto configured while I'm running `make`?

Yes they are.  You need to force the first one.  Try just removing the #if's, like this:
```c
//#if !defined(MICROPY_EMIT_THUMB) && defined(__thumb2__)
    #define MICROPY_EMIT_THUMB      (1)
    #define MICROPY_MAKE_POINTER_CALLABLE(p) ((void*)((mp_uint_t)(p) | 1))
//#endif
// Some compilers define __thumb2__ and __arm__ at the same time, let
// autodetected thumb2 emitter have priority.
//#if !defined(MICROPY_EMIT_ARM) && defined(__arm__) && !defined(__thumb2__)
//    #define MICROPY_EMIT_ARM        (1)
//#endif
```

COOLIRON2311, 2019-04-09:
The above programs now work fine. Thank you!

COOLIRON2311, 2019-04-10:
So, I assume that other decorators do not work and there is no way to make them work other than recompiling micropython without asm?

dpgeorge, 2019-04-11:
> So, I assume that the other decorators do not work now and there is no way to make all of them work at the same time?

Yes it's possible to make all decorators work at the same time.  It just needs some extra logic in the code to support Thumb inline assemble alongside normal ARM native code.

COOLIRON2311, 2019-04-12:
Will these changes be implemented at some point?

dpgeorge, 2019-05-01:
asm_thumb is now supported on ARM hosts, by 9ef784dcc6aa2abd7325b43f3d1613b8ac5d9f3f

For the unix port one will still need to explicitly enable it at build time via `make CFLAGS_EXTRA='-DMICROPY_EMIT_INLINE_THUMB=1'`

closed 2019-05-01 by dpgeorge

tompkinRV5, 2020-12-23:
@dpgeorge is it possible to add @micropython.asm_thumb support for a riscv arch. I am not sure if we should just add the arch(similar to how you solved it for arm) or maybe create a new decorator like @micropython.asm_riscv?