micropython/micropython #4678: STM32: No internal filesystem

mindforger, 2019-04-08:
i tried to open a file in the internal FS of my NUCLEO_L432KC

i searched issues far and wide but did not come up with what may be wrong with my compile

>$ os.stat("/")
(16384, 0, 0, 0, 0, 0, 0, 0, 0, 0)
>$ os.statvfs("/")
(0, 0, 0, 0, 0, 0, 0, 0, 0, 128)
>$ f = open("test.py","w")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 1] EPERM
>$ sys.path
['']

compile worked well and i deployed it using (after doing a mass erase with the original st-link tool from windows)
>make BOARD=NUCLEO_L432KC deploy-stlink

i am lost now as i have no clue how to get an SD Card running on bootup without main.py or boot.py

and sorry if i may have overseen the issue

dhylands, 2019-04-08:
The NUCLEO_L432KC only has 256K flash, so there isn't enough space for an internal file system.

You can copy your main.py and boot.py into the ports/stm32/modules directory and they'll get compiled into the firmware.

You also need to change main.c since the stm32 main.c doesn't currently support executing boot.py or main.py as frozen files.

For boot.py you want to make a change here:
https://github.com/micropython/micropython/blob/74ed06828f4f8d4ec5c04f5b551e90f2fccbd0f3/ports/stm32/main.c#L679-L691
```
    if (reset_mode == 1 || reset_mode == 3) {
        const char *boot_py = "boot.py";
#if 1
        pyexec_frozen_module(boot_py);
#else
        mp_import_stat_t stat = mp_import_stat(boot_py);
        if (stat == MP_IMPORT_STAT_FILE) {
            int ret = pyexec_file(boot_py);
            if (ret & PYEXEC_FORCED_EXIT) {
                goto soft_reset_exit;
            }
            if (!ret) {
                flash_error(4);
            }
        }
#endif
    }
```

And for main.py to want to make a similar change here:
https://github.com/micropython/micropython/blob/74ed06828f4f8d4ec5c04f5b551e90f2fccbd0f3/ports/stm32/main.c#L735-L752
```
    if ((reset_mode == 1 || reset_mode == 3) && pyexec_mode_kind == PYEXEC_MODE_FRIENDLY_REPL) {
        const char *main_py;
        if (MP_STATE_PORT(pyb_config_main) == MP_OBJ_NULL) {
            main_py = "main.py";
        } else {
            main_py = mp_obj_str_get_str(MP_STATE_PORT(pyb_config_main));
        }
#if 1
        pyexec_frozen_module(main_py);
#else
        mp_import_stat_t stat = mp_import_stat(main_py);
        if (stat == MP_IMPORT_STAT_FILE) {
            int ret = pyexec_file(main_py);
            if (ret & PYEXEC_FORCED_EXIT) {
                goto soft_reset_exit;
            }
            if (!ret) {
                flash_error(3);
            }
        }
#endif
    }
```
If you modify both, then you'll also want to comment out the soft_reset_exit label:
https://github.com/micropython/micropython/blob/74ed06828f4f8d4ec5c04f5b551e90f2fccbd0f3/ports/stm32/main.c#L768

It doesn't appear that FATFS support is included in the NUCLEO_L432KC, and there is only about 25K of unused flash space available, so you might not be able to add support for an SDCARD. I can't say for sure without having tried it.

dpgeorge, 2019-04-09:
I might be an interesting exercise to strip the configuration for this board down so that it does have enough room for a small internal filesystem.

> i had higher hopes for this tiny board (especially power wise)

Since this board is relatively new there's still a lot of potential to unlock from it.  And it's not that difficult, mostly a matter of configuration.  You should be able to get pretty low power consumption with machine.lightsleep() and machine.deepsleep().

dpgeorge, 2019-04-09:
I saved a bit of space in this board in 358364b45e21585103017325e72a36b2de4d2b6a .  It now has about 43000 bytes available.  Since it doesn't have USB the filesystem doesn't have to be FAT FS, it could be littlefs instead (although that's not fully integrated yet, so could use FAT FS for now).

mindforger, 2019-04-09:
wow thanks alot, i will try this out asap

but the other problem persists as i can do literally nothing to the board as a hit on tab-key after import only shows os and sys, but no machine or any peripherals

did i just misunderstand something, did i something wrong during compile or is there simply nothing implemented yet?

and i am pretty sure i can not use the sdcard.py without an working SPI interface :)

dhylands, 2019-04-09:
```
>>> help('modules')
```
will show you the available modules. On my NUCLEO_L432KC, I see the following:
```
MicroPython v1.10-260-g3dda96478-dirty on 2019-04-08; NUCLEO-L432KC with STM32L432KC
Type "help()" for more information.
>>> help('modules')
__main__          heapq             re                umachine
_onewire          io                select            uos
array             json              struct            urandom
binascii          lcd160cr          sys               ure
boot              lcd160cr_test     time              uselect
builtins          machine           ubinascii         ustruct
cmath             main              ucollections      utime
collections       math              uctypes           utimeq
dht               micropython       uerrno            uzlib
errno             onewire           uhashlib          zlib
framebuf          os                uheapq
gc                pyb               uio
hashlib           random            ujson
Plus any modules on the filesystem
>>> 
```

mindforger, 2019-04-10:
> help('modules')

didn't know that :) was used to type import and then hit tab

mhh .. do i actually require zlib, lcd160cr, json and framebuf? i think i might have to take a far deeper dive into how to configure included modules (makefiles are my biggest enemy)

my ordered raspi is coming today, so i can set it up as build system, because the VM i am using is repeatedly either BSOD my win 10 or it won't even start (curse you hyper-v) ... building native in win 10 with cross compiler always fails due to cmd lines generated from the makefile exceed 8k symbols

chrismas9, 2019-04-10:
re VM problems. Have you tried WSL on Windows 10? Just install Ubuntu from the Windows Store and run bash. It mounts your C drive ready to run make from your Windows file system. Has been very easy and reliable for me.

mindforger, 2019-04-10:
thx, will give it a shot :)

mindforger, 2019-04-18:
mhhh ... okay now i get it ... i can not use the STM32 sdcard.c because there is no SDMMC1 IRQ present on the L432KC .. took me a good while to understand all of those macros and relations

how can i "freeze" in the standard sdcard.py into my build so i can run the SPI version of that implementation? because dropping the sdcard.py into the modules folder did nothing for me

i could not use "import sdcard" and the build size did not increase wen dropping the file

dpgeorge, 2019-04-18:
@mindforger as you found out, there is no SDMMC peripheral on the L432.

> how can i "freeze" in the standard sdcard.py into my build so i can run the SPI version of that implementation? because dropping the sdcard.py into the modules folder did nothing for me

The default "modules" folder was disabled for this board to save flash.  You can build with your own custom modules directory via `make FROZEN_MPY_DIR=mymodules` (be sure to do `make clean` first).

dpgeorge, 2019-04-18:
@mindforger please see #4701 for changes to get a filesystem working on the L432.

mindforger, 2019-04-18:
holy crap, you must be reading minds(pun semi intended), i was just getting back to this :) will try out immediately

mindforger, 2019-04-18:
stupid question, how do i checkout that branch? never checked out a branch ever with git XD

mindforger, 2019-04-18:
> os.statvfs("/flash")
(512, 512, 18, 17, 17, 0, 0, 0, 0, 255)

briliant

closed 2019-04-18 by mindforger