micropython/micropython #4680: ESP32 Make Fails - pyparsing

pacmac, 2019-04-08:
```
GEN build/esp32.common.ld
Traceback (most recent call last):
  File "/nfs/qnap/data/uPython/build/micropython/ports/esp32/esp-idf/tools/ldgen/ldgen.py", line 22, in <module>
    from fragments import FragmentFileModel
  File "/nfs/qnap/data/uPython/build/micropython/ports/esp32/esp-idf/tools/ldgen/fragments.py", line 20, in <module>
    from sdkconfig import SDKConfig
  File "/nfs/qnap/data/uPython/build/micropython/ports/esp32/esp-idf/tools/ldgen/sdkconfig.py", line 18, in <module>
    from pyparsing import Word, printables, Combine, Literal, hexnums, quotedString, Optional, nums, removeQuotes, oneOf, Group, infixNotation, opAssoc
ModuleNotFoundError: No module named 'pyparsing'
Makefile:788: recipe for target 'build/esp32.common.ld' failed
make: *** [build/esp32.common.ld] Error 1
```

I fixed this by replacing every occurrence of $(PYTHON) with $(PYTHON2) in the Makefile.

pacmac, 2019-04-08:
I think this is caused because I have 3 versions of python:

```
update-alternatives --config python
There are 3 choices for the alternative python (providing /usr/bin/python).

  Selection    Path                      Priority   Status
------------------------------------------------------------
  0            /usr/local/bin/python3.7   2         auto mode
* 1            /usr/bin/python2.7         1         manual mode
  2            /usr/bin/python3.5         1         manual mode
  3            /usr/local/bin/python3.7   2         manual mode
```

dhylands, 2019-04-09:
You can also run[code]python -m pip install pyparsing[/code]to cover the case where pip didn't match up with the currently installed version of python. Replace python with python2 or 3 as appropriate.

dpgeorge, 2019-04-09:
Closing as a duplicate of #4655

closed 2019-04-09 by dpgeorge

pacmac, 2019-04-09:
OOps, sorrie about that.