micropython/micropython #4681: Pyboard D station interface: connection after power cycle unreliable

peterhinch, 2019-04-08:
Tests used an SF6W with a WBUS-DIP28. Firmware is the 14th April release build and mboot is updated. After each power down I waited 10s before reapplying power: this was sourced by a switched LiPo cell. No USB devices were connected. **boot.py** contained the correct country code only, and **main.py** was a one-liner importing the script below.

I also tested an SF2W with similar outcomes.

The tests were run in two locations. One was about 2.5 metres from the AP with an RSSI of -39dBm, the other in a room with RSSI of -58dBm. In the latter location ESP8266 and ESP32 devices connect with high reliability; the Pyboard D's can reconnect after a deliberate outage with 100% reliability in my testing. Results were as follows:

-39dBm location failure rate 3 out of 20 attempts.
-58dBm location failure rate 4 out of 5 attempts. Numerous tests with both Pyboard variants yielded a very low success rate.

The biggest problem is that if the initial connection fails, I have so far found no way to recover. Simply re-trying doesn't seem to work.
```python
import pyb, network, time
red = pyb.LED(1)
green = pyb.LED(2)

green.on()
time.sleep(4)  # Give things a chance to settle down
green.off()

s = network.WLAN(network.STA_IF)
s.active(True)
s.connect('redacted', 'REDACTED')
red.on()
time.sleep(0.2)  # Brief red flash if already connected
while not s.isconnected():
    time.sleep(1)
red.off()
green.on()
```
This was raised in [the forum](https://forum.micropython.org/viewtopic.php?f=20&t=6232&sid=bd54128309294199d2e67d038bec3d3a) where @chuckbook failed to replicate this, but it is a very real issue here.

chuckbook, 2019-04-08:
`
import pyb, network, time, machine

spin = '|/-\\'
red = pyb.LED(1)
green = pyb.LED(2)
blue = pyb.LED(3)

green.on()
time.sleep(4)  # Give things a chance to settle down
green.off()
blue.off()

s = network.WLAN(network.STA_IF)
s.active(True)
s.connect('redacted', 'redacted')
red.on()
time.sleep(0.2)  # Brief red flash if already connected
trial = 100
while True:
    if s.isconnected():
        red.off()
        blue.off()
        green.on()
        break
    time.sleep(0.1)
    trial -= 1
    red.toggle()
    blue.toggle()
    print(end='%c\x08' % (spin[trial % len(spin)]))
    if trial <= 0:
        red.on()
        green.off()
        blue.off()
        machine.soft_reset()
        break;

print()
`

peterhinch, 2019-04-08:
OK, thanks: we have a workround, albeit one which is a bit drastic.

dpgeorge, 2019-04-09:
Please try to enable async event tracing via:
```python
s.config(trace=1)
```
That will print logs of all WiFi related events.  There are a certain set of events that are needed to form a connection, and in your case it may be that only a few are there.  If you can report the events for successful and failed connections that would be very helpful.

Also, it's possible to completely turn off the WLAN chip via `s.deinit()`, then reinit via `s.active(1)`.

peterhinch, 2019-04-09:
To monitor this I used uos.dupterm with an FTDI connection on UART1.

Here is the log of a failed connection from the application of power to the board.
```
[adminpete@capybara]: ~
$ mtftdi
--- Miniterm on /dev/ftdi  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
[    7238] ASYNC(0000,ASSOC_REQ_IE,0,0,0,77,00:00:00:00:00:00,wl0,0,0)
[    7245] ASYNC(0000,AUTH,0,0,0,0,00:1d:aa:84:24:d0,wl0,0,0)
[    7251] ASYNC(0000,ASSOC_RESP_IE,0,0,0,51,00:00:00:00:00:00,wl0,0,0)
[    7257] ASYNC(0000,ASSOC,0,0,0,57,00:1d:aa:84:24:d0,wl0,0,0)
[    7263] ASYNC(0001,LINK,0,0,0,22,00:1d:aa:84:24:d0,wl0,0,0)
[    7333] ASYNC(0000,JOIN,0,0,0,0,00:1d:aa:84:24:d0,wl0,0,0)
[    7339] ASYNC(0000,SET_SSID,0,0,0,9,00:1d:aa:84:24:d0,wl0,0,0)
[    9741] ASYNC(0000,PSK_SUP,4,15,0,0,00:1d:aa:84:24:d0,wl0,0,0)
[   11330] ASYNC(0000,51,0,0,0,0,00:1d:aa:84:24:d0,wl0,0,0)
[   12354] ASYNC(0000,51,0,0,0,0,00:1d:aa:84:24:d0,wl0,0,0)
[   14084] ASYNC(0000,PSK_SUP,4,14,0,0,00:00:00:00:00:00,wl0,0,0)
[   14090] ASYNC(0000,DEAUTH_IND,0,15,0,2,00:1d:aa:84:24:d0,wl0,0,0)
[   14096] ASYNC(0000,67,0,0,0,292,00:1d:aa:84:24:d0,wl0,0,0)
[   14483] ASYNC(0000,32,0,0,0,0,00:1d:aa:84:24:d0,wl0,0,0)
[   14489] ASYNC(0000,ASSOC_REQ_IE,0,0,0,77,00:00:00:00:00:00,wl0,0,0)
[   14495] ASYNC(0000,AUTH,0,0,0,0,00:1d:aa:84:24:d0,wl0,0,0)
[   14501] ASYNC(0000,ASSOC_RESP_IE,0,0,0,51,01:10:10:44:00:01,wl0,0,0)
[   14507] ASYNC(0000,9,0,0,0,57,00:1d:aa:84:24:d0,wl0,0,0)
[   14513] ASYNC(0001,LINK,0,0,0,22,00:1d:aa:84:24:d0,wl0,0,0)
[   14519] ASYNC(0000,JOIN,0,0,0,0,00:1d:aa:84:24:d0,wl0,0,0)
[   14524] ASYNC(0000,125,0,0,0,0,00:1d:aa:84:24:d0,wl0,0,0)
[   15225] ASYNC(0000,PSK_SUP,6,0,0,0,00:1d:aa:84:24:d0,wl0,0,0)
```

peterhinch, 2019-04-09:
> Also, it's possible to completely turn off the WLAN chip via s.deinit(), then reinit via s.active(1).

That seems to provide a good workround. Here is my modified script and the trace output from a connect which failed on the first pass but worked on the second. So far it's always connected on the second pass. I should stress that this issue only seems to occur after a complete power outage.
```python
import pyb, network, time
red = pyb.LED(1)
green = pyb.LED(2)

green.on()
time.sleep(4)  # Give things a chance to settle down
green.off()

s = network.WLAN(network.STA_IF)
s.config(trace=1)
while True:
    s.active(True)
    s.connect('redacted', 'redacted')
    red.on()
    time.sleep(0.2)  # Brief red flash if already connected
    for _ in range(10):
        if s.isconnected():
            break
        time.sleep(1)
    else:
        s.deinit()
        print('Retry connection')
        red.off()
        time.sleep(0.5)
        red.on()
        continue
    print('Connected')
    break
red.off()
green.on()
```
Output on power up via FTDI adapter on UART1:
```
MicroPython v1.10-443-g4462ed4e2-dirty on 2019-04-02; PYBD_SF6W with STM32F767IIK
Type "help()" for more information.
>>> [    7240] ASYNC(0000,ASSOC_REQ_IE,0,0,0,77,00:00:00:00:00:00,wl0,0,0)
[    7246] ASYNC(0000,AUTH,0,0,0,0,00:1d:aa:84:24:d0,wl0,0,0)
[    7252] ASYNC(0000,ASSOC_RESP_IE,0,0,0,51,00:00:00:00:00:00,wl0,0,0)
[    7259] ASYNC(0000,ASSOC,0,0,0,57,00:1d:aa:84:24:d0,wl0,0,0)
[    7265] ASYNC(0001,LINK,0,0,0,22,00:1d:aa:84:24:d0,wl0,0,0)
[    7270] ASYNC(0000,JOIN,0,0,0,0,00:1d:aa:84:24:d0,wl0,0,0)
[    7276] ASYNC(0000,SET_SSID,0,0,0,9,00:1d:aa:84:24:d0,wl0,0,0)
[    7971] ASYNC(0000,51,0,0,0,0,00:1d:aa:84:24:d0,wl0,0,0)
[    9742] ASYNC(0000,PSK_SUP,4,15,0,0,00:1d:aa:84:24:d0,wl0,0,0)
[   10019] ASYNC(0000,51,0,0,0,0,00:1d:aa:84:24:d0,wl0,0,0)
[   13611] ASYNC(0000,PSK_SUP,4,14,0,0,00:00:00:00:00:00,wl0,0,0)
[   13617] ASYNC(0000,DEAUTH_IND,0,15,0,2,00:1d:aa:84:24:d0,wl0,0,0)
[   13623] ASYNC(0000,67,0,0,0,292,00:1d:aa:84:24:d0,wl0,0,0)
[   14009] ASYNC(0000,32,0,0,0,0,00:1d:aa:84:24:d0,wl0,0,0)
[   14014] ASYNC(0000,ASSOC_REQ_IE,0,0,0,77,00:00:ac:16:65:23,wl0,0,0)
[   14021] ASYNC(0000,AUTH,0,0,0,0,00:1d:aa:84:24:d0,wl0,0,0)
[   14026] ASYNC(0000,ASSOC_RESP_IE,0,0,0,51,05:00:00:00:00:00,wl0,0,0)
[   14033] ASYNC(0000,9,0,0,0,57,00:1d:aa:84:24:d0,wl0,0,0)
[   14038] ASYNC(0001,LINK,0,0,0,22,00:1d:aa:84:24:d0,wl0,0,0)
[   14112] ASYNC(0000,JOIN,0,0,0,0,00:1d:aa:84:24:d0,wl0,0,0)
[   14118] ASYNC(0000,125,0,0,0,0,00:1d:aa:84:24:d0,wl0,0,0)
Retry connection
[   18350] ASYNC(0000,ASSOC_REQ_IE,0,0,0,77,00:00:00:00:00:00,wl0,0,0)
[   18357] ASYNC(0000,AUTH,0,0,0,0,00:1d:aa:84:24:d0,wl0,0,0)
[   18362] ASYNC(0000,ASSOC_RESP_IE,0,0,0,51,00:00:00:00:00:00,wl0,0,0)
[   18369] ASYNC(0000,ASSOC,0,0,0,57,00:1d:aa:84:24:d0,wl0,0,0)
[   18375] ASYNC(0001,LINK,0,0,0,22,00:1d:aa:84:24:d0,wl0,0,0)
[   18413] ASYNC(0000,JOIN,0,0,0,0,00:1d:aa:84:24:d0,wl0,0,0)
[   18419] ASYNC(0000,SET_SSID,0,0,0,9,00:1d:aa:84:24:d0,wl0,0,0)
[   19557] ASYNC(0000,PSK_SUP,6,0,0,0,00:00:00:00:20:00,wl0,0,0)
Connected
MicroPython v1.10-443-g4462ed4e2-dirty on 2019-04-02; PYBD_SF6W with STM32F767IIK
Type "help()" for more information.
>>> 
```

peterhinch, 2019-04-09:
I amended my script to use status == 1.
```python
import pyb, network, time
red = pyb.LED(1)
green = pyb.LED(2)

green.on()
time.sleep(4)  # Give things a chance to settle down
green.off()

s = network.WLAN(network.STA_IF)
s.config(trace=1)
while True:
    s.active(True)
    s.connect('redacted', 'redacted')
    red.on()
    time.sleep(0.2)  # Brief red flash if already connected
    while s.status() == 1:
        time.sleep(1)
    if s.isconnected():
        break
    s.deinit()
    print('Retry connection')
    red.off()
    time.sleep(0.5)
    red.on()
print('Connected')
red.off()
green.on()
```
Here is the outcome when it only connected on the third pass:
```
MicroPython v1.10-443-g4462ed4e2-dirty on 2019-04-02; PYBD_SF6W with STM32F767IIK
Type "help()" for more information.
>>> [    7240] ASYNC(0000,ASSOC_REQ_IE,0,0,0,77,00:00:00:00:00:00,wl0,0,0)
[    7246] ASYNC(0000,AUTH,0,0,0,0,00:1d:aa:84:24:d0,wl0,0,0)
[    7252] ASYNC(0000,ASSOC_RESP_IE,0,0,0,51,00:00:00:00:00:00,wl0,0,0)
[    7259] ASYNC(0000,ASSOC,0,0,0,57,00:1d:aa:84:24:d0,wl0,0,0)
[    7265] ASYNC(0001,LINK,0,0,0,22,00:1d:aa:84:24:d0,wl0,0,0)
[    7270] ASYNC(0000,JOIN,0,0,0,0,00:1d:aa:84:24:d0,wl0,0,0)
[    7276] ASYNC(0000,SET_SSID,0,0,0,9,00:1d:aa:84:24:d0,wl0,0,0)
[    9742] ASYNC(0000,PSK_SUP,4,15,0,0,00:1d:aa:84:24:d0,wl0,0,0)
Retry connection
[   14351] ASYNC(0000,ASSOC_REQ_IE,0,0,0,77,00:00:00:00:00:00,wl0,0,0)
[   14357] ASYNC(0000,AUTH,0,0,0,0,00:1d:aa:84:24:d0,wl0,0,0)
[   14363] ASYNC(0000,ASSOC_RESP_IE,0,0,0,51,00:00:00:00:00:00,wl0,0,0)
[   14370] ASYNC(0000,ASSOC,0,0,0,57,00:1d:aa:84:24:d0,wl0,0,0)
[   14375] ASYNC(0001,LINK,0,0,0,22,00:1d:aa:84:24:d0,wl0,0,0)
[   14434] ASYNC(0000,JOIN,0,0,0,0,00:1d:aa:84:24:d0,wl0,0,0)
[   14439] ASYNC(0000,SET_SSID,0,0,0,9,00:1d:aa:84:24:d0,wl0,0,0)
[   16855] ASYNC(0000,PSK_SUP,4,15,0,0,00:1d:aa:84:24:d0,wl0,0,0)
Retry connection
[   21458] ASYNC(0000,ASSOC_REQ_IE,0,0,0,77,00:00:00:00:00:00,wl0,0,0)
[   21464] ASYNC(0000,AUTH,0,0,0,0,00:1d:aa:84:24:d0,wl0,0,0)
[   21470] ASYNC(0000,ASSOC_RESP_IE,0,0,0,51,00:00:00:00:00:00,wl0,0,0)
[   21477] ASYNC(0000,ASSOC,0,0,0,57,00:1d:aa:84:24:d0,wl0,0,0)
[   21483] ASYNC(0001,LINK,0,0,0,22,00:1d:aa:84:24:d0,wl0,0,0)
[   21499] ASYNC(0000,JOIN,0,0,0,0,00:1d:aa:84:24:d0,wl0,0,0)
[   21505] ASYNC(0000,SET_SSID,0,0,0,9,00:1d:aa:84:24:d0,wl0,0,0)
[   21749] ASYNC(0000,PSK_SUP,6,0,0,0,00:0c:43:00:00:00,wl0,0,0)
Connected
MicroPython v1.10-443-g4462ed4e2-dirty on 2019-04-02; PYBD_SF6W with STM32F767IIK
Type "help()" for more information.
>>> 
```

peterhinch, 2019-04-09:
Another wrinkle. In my resilient MQTT application if a WiFi outage occurs the code issues the station interface **disconnect()** method then attempts to reconnect. This is in a loop which only breaks out on success. This works.

Experimentally after the **disconnect()** I tried doing **deinit()** with the following outcome on a temporary WiFi failure:
```
WiFi or broker is down.
bcm43x_kso_set(1): failed
sdio_transfer_cmd53: error=00082004 wr=0 len=4 dma=0 buf_idx=0 STA=00082000 SDMMC=00000004:00000001 DMA=00000000:00000000:00000000 RCC=0010017f
sdio_transfer_cmd53: error=00082004 wr=0 len=4 dma=0 buf_idx=0 STA=00082000 SDMMC=00000004:00000001 DMA=00000000:00000000:00000000 RCC=0010017f
[CYW43] error: hdr mismatch ed05 ^ 0804
Checking WiFi integrity.
Got reliable connection
```
As you can see the application continued to run, but I have reverted the **deinit()** to avoid the spam as **deinit()** is unnecessary on a reconnection. This is not actually a problem to me but perhaps the information is useful.

dpgeorge, 2019-04-09:
> ASYNC(0000,PSK_SUP,4,15,0,0)

This means the 4-way WPA handshake timed out, most likely due to lost packets.

What kind of security does your AP use?  If you do a `wl.scan()` it's the second last number in the tuple.  It should be 0, 1, 2 or 3.

peterhinch, 2019-04-09:
`(b'redacted', b'\x00\x1d\xaa\x84$\xd0', 3, -61, 3, 7)`

dpgeorge, 2019-04-11:
Ok, thanks for the info.  I did manage to see the WPA handshake timeout once in my set up and think I have enough info now to improve it.  Basically, this event should be handled by the low-level driver and the connection automatically retried.

dpgeorge, 2019-04-11:
I push a commit to the stm32-pybd branch which might help to fix the issue here of poor connection.  Please can you test it out by checking out the branch and building it?

peterhinch, 2019-04-12:
A major improvement. In the -58dBm location the failure rate was 4 in 20 attempts (down from 4 in 5).

peterhinch, 2019-04-12:
My concern is that there seems to be no way to retry a connection. There will always be cases where an initial connection fails, e.g. due to RFI. Further there is bound to be an RSSI  level where connection may sometimes fail.

The following attempts to reconnect but seems unable to do so:
```python
import pyb, network, time
red = pyb.LED(1)
green = pyb.LED(2)

green.on()
time.sleep(4)  # Give things a chance to settle down
green.off()

s = network.WLAN(network.STA_IF)
s.config(trace=1)
while True:
    s.active(True)
    s.connect('ssid', 'pw')
    red.on()
    time.sleep(0.2)  # Brief red flash if already connected
    while s.status() == 1:
        time.sleep(1)
    if s.isconnected():
        break
    s.deinit()
    print('Retry connection')
    red.off()
    time.sleep(0.5)
    red.on()
print('Connected')
red.off()
green.on()
```
On failure the red LED remains lit permanently (no 0.5s blinks, no 'Retry connection' output) and it produces the following (forever)
```
>>> [    7253] ASYNC(0000,ASSOC_REQ_IE,0,0,0)
[    7257] ASYNC(0000,AUTH,0,0,0)
[    7260] ASYNC(0000,ASSOC_RESP_IE,0,0,0)
[    7264] ASYNC(0000,ASSOC,0,0,0)
[    7267] ASYNC(0001,LINK,0,0,0)
[    7302] ASYNC(0000,JOIN,0,0,0)
[    7305] ASYNC(0000,SET_SSID,0,0,0)
[    9757] ASYNC(0000,PSK_SUP,4,15,0)
[   10162] ASYNC(0000,ASSOC_REQ_IE,0,0,0)
[   10166] ASYNC(0000,AUTH,0,0,0)
[   10169] ASYNC(0000,ASSOC_RESP_IE,0,0,0)
[   10173] ASYNC(0000,ASSOC,0,0,0)
[   10176] ASYNC(0001,LINK,0,0,0)
[   10179] ASYNC(0000,JOIN,0,0,0)
[   10182] ASYNC(0000,SET_SSID,0,0,0)
[   12667] ASYNC(0000,PSK_SUP,4,15,0)
[   13068] ASYNC(0000,ASSOC_REQ_IE,0,0,0)
[   13072] ASYNC(0000,AUTH,0,0,0)
[   13075] ASYNC(0000,ASSOC_RESP_IE,0,0,0)
[   13079] ASYNC(0000,ASSOC,0,0,0)
[   13082] ASYNC(0001,LINK,0,0,0)
[   13139] ASYNC(0000,JOIN,0,0,0)
[   13142] ASYNC(0000,SET_SSID,0,0,0)
[   15572] ASYNC(0000,PSK_SUP,4,15,0)
[   15975] ASYNC(0000,ASSOC_REQ_IE,0,0,0)
[   15979] ASYNC(0000,AUTH,0,0,0)
[   15982] ASYNC(0000,ASSOC_RESP_IE,0,0,0)
[   15986] ASYNC(0000,ASSOC,0,0,0)
[   15989] ASYNC(0001,LINK,0,0,0)
[   16006] ASYNC(0000,JOIN,0,0,0)
[   16009] ASYNC(0000,SET_SSID,0,0,0)
[   18478] ASYNC(0000,PSK_SUP,4,15,0)
[   18882] ASYNC(0000,ASSOC_REQ_IE,0,0,0)
[   18886] ASYNC(0000,AUTH,0,0,0)
[   18889] ASYNC(0000,ASSOC_RESP_IE,0,0,0)
[   18893] ASYNC(0000,ASSOC,0,0,0)
[   18896] ASYNC(0001,LINK,0,0,0)
[   18976] ASYNC(0000,JOIN,0,0,0)
[   18979] ASYNC(0000,SET_SSID,0,0,0)
[   21386] ASYNC(0000,PSK_SUP,4,15,0)
[   21791] ASYNC(0000,ASSOC_REQ_IE,0,0,0)
[   21795] ASYNC(0000,AUTH,0,0,0)
[   21798] ASYNC(0000,ASSOC_RESP_IE,0,0,0)
[   21802] ASYNC(0000,ASSOC,0,0,0)
[   21805] ASYNC(0001,LINK,0,0,0)
[   21843] ASYNC(0000,JOIN,0,0,0)
[   21846] ASYNC(0000,SET_SSID,0,0,0)
[   24294] ASYNC(0000,PSK_SUP,4,15,0)
[   24698] ASYNC(0000,ASSOC_REQ_IE,0,0,0)
[   24702] ASYNC(0000,AUTH,0,0,0)
[   24705] ASYNC(0000,ASSOC_RESP_IE,0,0,0)
[   24709] ASYNC(0000,ASSOC,0,0,0)
[   24712] ASYNC(0001,LINK,0,0,0)
[   24715] ASYNC(0000,JOIN,0,0,0)
[   24718] ASYNC(0000,SET_SSID,0,0,0)
[   27201] ASYNC(0000,PSK_SUP,4,15,0)
[   27604] ASYNC(0000,ASSOC_REQ_IE,0,0,0)
[   27608] ASYNC(0000,AUTH,0,0,0)
[   27611] ASYNC(0000,ASSOC_RESP_IE,0,0,0)
[   27615] ASYNC(0000,ASSOC,0,0,0)
[   27618] ASYNC(0001,LINK,0,0,0)
[   27680] ASYNC(0000,JOIN,0,0,0)
[   27683] ASYNC(0000,SET_SSID,0,0,0)
[   30108] ASYNC(0000,PSK_SUP,4,15,0)
[   30510] ASYNC(0000,ASSOC_REQ_IE,0,0,0)
[   30514] ASYNC(0000,AUTH,0,0,0)
....
```

dpgeorge, 2019-04-12:
Thanks for testing it.

If you put a simple timeout in the waiting part, does that stop the infinite retry and get it to eventually connect?  That is:
```python
    ...
    t0 = time.ticks_ms()
    while s.status() == 1 and time.ticks_diff(time.ticks_ms(), t0) < 10000:
        time.sleep(1)
    ...
```

peterhinch, 2019-04-12:
No. When it fails it keeps re-trying every 10s indefinitely.

peterhinch, 2019-04-26:
A further possible manifestation of this problem - or perhaps something different. The following script runs for a while (~30 mins) before failing to connect:
```python
from umqtt.simple import MQTTClient
import network
import ujson
from  local import SERVER, SSID, PW  # Local configuration: change this file
import uasyncio as asyncio

def publish(s):
    c = MQTTClient('umqtt_client', SERVER)
    c.connect()
    c.publish(b'result', s.encode('UTF8'))
    c.disconnect()

async def wifi_connect(sta_if, red, grn):
    while True:
        red.on()  # Connecting
        grn.off()
        sta_if.active(True)
        sta_if.connect(SSID, PW)
        while sta_if.status() in (1, 2):  # https://github.com/micropython/micropython/issues/4682
            await asyncio.sleep(1)
            grn.toggle()  # Loops here forever when it fails
        if sta_if.isconnected():
            red.off()
            grn.on()
            await asyncio.sleep(1)  # 1s of green == success.
            grn.off() 
            break
        sta_if.deinit()  # Turn off WiFi chip
        await asyncio.sleep(1)  # Try to reconnect

async def main(loop):
    rtc = RTC()
    red = LED(1)
    grn = LED(2)
    count = 0
    sta_if = network.WLAN()
    sta_if.config(trace=1)
    while True:
        await wifi_connect(sta_if, red, grn)
        publish(ujson.dumps([count, rtc.datetime()]))
        sta_if.deinit()  # Turn off WiFi chip
        count += 1
        await asyncio.sleep(10)

loop = asyncio.get_event_loop()
loop.run_until_complete(main(loop))
```
The odd thing is that the trace output on failure looks the same as for prior successful connects
```
sdio_transfer_cmd53: timeout wr=0 len=4 dma=0 buf_idx=0 STA=00000000 SDMMC=00000000:00000000 DMA=00000000:00000000:00000000 RCC=2010017f
sdio_transfer_cmd53: timeout wr=0 len=4 dma=0 buf_idx=0 STA=00000000 SDMMC=00000000:00000000 DMA=00000000:00000000:00000000 RCC=2010017f
[CYW43] error: hdr mismatch a891 ^ 0804
[ 1548977] ASYNC(0000,ASSOC_REQ_IE,0,0,0)
[ 1548977] ASYNC(0000,AUTH,0,0,0)
[ 1548982] ASYNC(0000,ASSOC_RESP_IE,0,0,0)
[ 1548982] ASYNC(0000,ASSOC,0,0,0)
[ 1548982] ASYNC(0001,LINK,0,0,0)
[ 1548992] ASYNC(0000,JOIN,0,0,0)
[ 1548992] ASYNC(0000,SET_SSID,0,0,0)
[ 1549327] ASYNC(0000,PSK_SUP,6,0,0)
sdio_transfer_cmd53: timeout wr=0 len=4 dma=0 buf_idx=0 STA=00000000 SDMMC=00000000:00000000 DMA=00000000:00000000:00000000 RCC=2010017f
sdio_transfer_cmd53: timeout wr=0 len=4 dma=0 buf_idx=0 STA=00000000 SDMMC=00000000:00000000 DMA=00000000:00000000:00000000 RCC=2010017f
[CYW43] error: hdr mismatch a891 ^ 0804
[ 1565237] ASYNC(0000,ASSOC_REQ_IE,0,0,0)
[ 1565237] ASYNC(0000,AUTH,0,0,0)
[ 1565241] ASYNC(0000,ASSOC_RESP_IE,0,0,0)
[ 1565241] ASYNC(0000,ASSOC,0,0,0)
[ 1565241] ASYNC(0001,LINK,0,0,0)
[ 1565274] ASYNC(0000,JOIN,0,0,0)
[ 1565274] ASYNC(0000,SET_SSID,0,0,0)
[ 1565506] ASYNC(0000,PSK_SUP,6,0,0)
sdio_transfer_cmd53: timeout wr=0 len=4 dma=0 buf_idx=0 STA=00000000 SDMMC=00000000:00000000 DMA=00000000:00000000:00000000 RCC=2010017f
sdio_transfer_cmd53: timeout wr=0 len=4 dma=0 buf_idx=0 STA=00000000 SDMMC=00000000:00000000 DMA=00000000:00000000:00000000 RCC=2010017f
[CYW43] error: hdr mismatch a891 ^ 0804
[ 1581498] ASYNC(0000,ASSOC_REQ_IE,0,0,0)
[ 1581499] ASYNC(0000,AUTH,0,0,0)
[ 1581504] ASYNC(0000,ASSOC_RESP_IE,0,0,0)
[ 1581504] ASYNC(0000,ASSOC,0,0,0)
[ 1581505] ASYNC(0001,LINK,0,0,0)
[ 1581556] ASYNC(0000,JOIN,0,0,0)
[ 1581556] ASYNC(0000,SET_SSID,0,0,0)
[ 1581788] ASYNC(0000,PSK_SUP,6,0,0)
[hangs forever here]
```
Trace shows last 3 successful publications. No more trace output occurs but the red LED is on and the green is toggling once per second.

dpgeorge, 2019-04-30:
Thanks @peterhinch for that new test.  I can reproduce the same problem, and it always happens on the 95th connection.  The issue is that the lwIP stack runs out of memory and the DHCP client cannot allocate memory to send out a DHCP message to request an IP address, so it gets stuck waiting forever for one.

So there's a lwIP-related memory leak somewhere.

dpgeorge, 2019-04-30:
> So there's a lwIP-related memory leak somewhere.

It turned out to be mDNS which was not being deinitialised when `wl.deinit()` was called.  Also the DHCP client and server were retaining some UDP memory (but that didn't manifest itself in the above tests).  These issues are all fixed by the latest commits on the stm32-pybd branch.

I still think there's an outstanding issue with `wl.connect()` looping forever.

peterhinch, 2019-05-16:
Re the original issue, connection after power up is much improved. But I'm seeing a state where repeated connection attempts fail. The easiest way to provoke this is to power the board up in a location where the signal strength is marginal. If it repeatedly fails to connect, this behaviour sometimes persists even if you move the board and its power source close to the AP.

Script invoked by main.py:
```python
import pyb, network, time
red = pyb.LED(1)
green = pyb.LED(2)

green.on()
time.sleep(4)  # Give things a chance to settle down
green.off()

s = network.WLAN(network.STA_IF)
s.config(trace=1)
while True:
    s.active(True)
    s.connect('ssid', 'pw')
    red.on()
    time.sleep(0.2)  # Brief red flash if already connected
    t0 = time.ticks_ms()
    while s.status() in (1, 2) and time.ticks_diff(time.ticks_ms(), t0) < 10000:
        time.sleep(1)
    if s.isconnected():
        break
    s.deinit()
    print('Retry connection')
    red.off()
    time.sleep(0.5)
    red.on()
print('Connected')
red.off()
green.on()
```
Firmware downloaded today
```
MicroPython v1.10-347-ga13e5862c on 2019-04-30; PYBD_SF6W with STM32F767IIK
```
A sample of the trace output. This was captured after moving close to the AP (RSSI -39dBm).
```
...
[   79843] ASYNC(0000,JOIN,0,0,0)
[   79843] ASYNC(0000,SET_SSID,0,0,0)
Retry connection
[   85058] ASYNC(0000,ASSOC_REQ_IE,0,0,0)
[   85058] ASYNC(0000,AUTH,0,0,0)
[   85061] ASYNC(0000,ASSOC_RESP_IE,0,0,0)
[   85061] ASYNC(0000,ASSOC,0,0,0)
[   85061] ASYNC(0001,LINK,0,0,0)
[   85066] ASYNC(0000,JOIN,0,0,0)
[   85066] ASYNC(0000,SET_SSID,0,0,0)
[   87561] ASYNC(0000,PSK_SUP,4,15,0)
[   87963] ASYNC(0000,ASSOC_REQ_IE,0,0,0)
[   87963] ASYNC(0000,AUTH,0,0,0)
[   87966] ASYNC(0000,ASSOC_RESP_IE,0,0,0)
[   87966] ASYNC(0000,ASSOC,0,0,0)
[   87966] ASYNC(0001,LINK,0,0,0)
[   88035] ASYNC(0000,JOIN,0,0,0)
[   88035] ASYNC(0000,SET_SSID,0,0,0)
[   90466] ASYNC(0000,PSK_SUP,4,15,0)
[   90868] ASYNC(0000,ASSOC_REQ_IE,0,0,0)
[   90868] ASYNC(0000,AUTH,0,0,0)
[   90871] ASYNC(0000,ASSOC_RESP_IE,0,0,0)
[   90871] ASYNC(0000,ASSOC,0,0,0)
[   90871] ASYNC(0001,LINK,0,0,0)
[   90902] ASYNC(0000,JOIN,0,0,0)
[   90903] ASYNC(0000,SET_SSID,0,0,0)
Retry connection
[   96173] ASYNC(0000,ASSOC_REQ_IE,0,0,0)
[   96173] ASYNC(0000,AUTH,0,0,0)
[   96176] ASYNC(0000,ASSOC_RESP_IE,0,0,0)
[   96176] ASYNC(0000,ASSOC,0,0,0)
[   96176] ASYNC(0001,LINK,0,0,0)
[   96227] ASYNC(0000,JOIN,0,0,0)
[   96228] ASYNC(0000,SET_SSID,0,0,0)
[   98676] ASYNC(0000,PSK_SUP,4,15,0)
[   99075] ASYNC(0000,ASSOC_REQ_IE,0,0,0)
[   99075] ASYNC(0000,AUTH,0,0,0)
[   99078] ASYNC(0000,ASSOC_RESP_IE,0,0,0)
[   99078] ASYNC(0000,ASSOC,0,0,0)
[   99078] ASYNC(0001,LINK,0,0,0)
[   99095] ASYNC(0000,JOIN,0,0,0)
[   99095] ASYNC(0000,SET_SSID,0,0,0)
[  101579] ASYNC(0000,PSK_SUP,4,15,0)
[  101979] ASYNC(0000,ASSOC_REQ_IE,0,0,0)
[  101979] ASYNC(0000,AUTH,0,0,0)
[  101982] ASYNC(0000,ASSOC_RESP_IE,0,0,0)
[  101982] ASYNC(0000,ASSOC,0,0,0)
[  101982] ASYNC(0001,LINK,0,0,0)
[  102064] ASYNC(0000,JOIN,0,0,0)
[  102064] ASYNC(0000,SET_SSID,0,0,0)
Retry connection
[  107292] ASYNC(0000,ASSOC_REQ_IE,0,0,0)
...
```

dpgeorge, 2019-05-17:
Thanks for the test, I'll look into it.

peterhinch, 2019-07-26:
Any progress on this one? The bug is fatal for applications which must recover from a WiFi outage without losing state.

dpgeorge, 2019-07-31:
I have tried but I'm unable to reproduce the issue.  Using the uFL/external antenna without an antenna connected (via `wl.config(antenna=1)`) lets me get very poor reception (eg -94dBm) but it's still able to eventually connect.

What authentication is your router set to use (eg WPA2 only)?

peterhinch, 2019-07-31:
I'm sorry, I hadn't realised you hadn't replicated this. From my router security setup screen:
```
Mode: WPA2/PSK		
WPA:
	Encryption Mode:	TKIP for WPA/AES for WPA2
	Pre-Shared Key(PSK):	*************

 Type 8~63 ASCII character or 64 Hexadecimal digits leading by "0x",
 for example "cfgs01a2..." or "0x655abcd....".
```
I guess I must have entered a PSK when I first got the router a few years ago.

dpgeorge, 2019-08-01:
Your AP supports WPA with TKIP, and WPA2 with AES.  The PYBD attempts to connect with WPA2 TKIP or WPA2 AES.  And from the ASYNC logs above the `ASYNC(0000,PSK_SUP,4,15,0)` event (a handshake timeout) is occurring 2500ms (the default timeout) after the `ASYNC(0000,ASSOC,0,0,0)` event.

Can you please try forcing the authentication to a specific type, by using one of the following:
```
s.connect('ssid', 'pw', auth=0x0402) # WPA with TKIP <- try this
s.connect('ssid', 'pw', auth=0x0404) # WPA with AES
s.connect('ssid', 'pw', auth=0x8002) # WPA2 with TKIP
s.connect('ssid', 'pw', auth=0x8004) # WPA2 with AES <- try this
s.connect('ssid', 'pw', auth=0x8006) # WPA2 with TKIP/AES (the default)
```

peterhinch, 2019-08-01:
Testing against ```MicroPython v1.11-187-g00e7fe8ab```, I have so far been unable to replicate the fault with the script as posted. Have there been improvements to the firmware which may have fixed this fault? If so, maybe we can close this. Otherwise I'll do more testing.

dpgeorge, 2019-08-02:
>  Have there been improvements to the firmware which may have fixed this fault?

Yes there have, but only to the generic stm32 code, nothing really PYBD specific.  If you could do a bit more testing that would be great.

peterhinch, 2019-08-02:
In further testing using firmware built from source I still can't replicate the fault. Yet it was easily provoked before.

The other change is hardware: my original testing was using the boards you gave me. That SF2W is now built in to my thermal IR camera. I'm testing with an SF2W from GR delivered last week. Have there been relevant hardware changes?

dpgeorge, 2019-08-02:
>  Have there been relevant hardware changes?

No there shouldn't have been anything.

To narrow the problem down, find here the same firmware reported above that showed the issue: https://micropython.org/resources/firmware/PYBD_SF2W_v1.10-347-ga13e5862c.dfu
Could you please test that?

peterhinch, 2019-08-02:
This is where it gets rather subjective. tl;dr I can't replicate the original result with this firmware.

It seems more reluctant to connect under low RSSI conditions than current firmware. In the original result I could take the board within 2 metres of the AP and it still wouldn't connect. Now, in every test it connected by the time I approached the house front door, ~9m from the AP through a closed door.

In tests with the new firmware it always connected while I was some way down the road, >= 30m from the AP.

peterhinch, 2019-08-05:
I have repeated the test with the original SF2W board and have replicated the fault, twice with the original firmware and once with firmware built from the latest source. Despite many attempts I never replicated it with the new board.

Interestingly in the failing case, with the board 2m from the AP, it does eventually connect, but you can have a long wait (minutes).

The behaviour is so non-deterministic that it's hard to draw reliable conclusions: even the old board usually connects at long range. But it does seem to be hardware related. I'm not sure where we go from here: if neither of us can replicate it with current hardware I guess it can be closed.

dpgeorge, 2019-08-06:
@peterhinch thanks for all the testing.  It sounds like it could be hardware related.  Can you please email me a photo of the top of the board that is showing the problems?

peterhinch, 2019-08-06:
YHM.