micropython/micropython #4682: Pyboard D network module station interface: please implement STAT_CONNECTING

peterhinch, 2019-04-08:
When calling the station interface **connect** method there is a need to verify whether the connection succeeded. On ESP8266 and ESP32 this can be done with
```python
while s.status() == network.STAT_CONNECTING:  # Break out on fail or success.
    await asyncio.sleep(1)
if s.isconnected():
    #
```
Currently the only way seems to be to poll **s.isconnected()** with an arbitrary timeout.

dpgeorge, 2019-04-09:
`s.status()` should be returning 1 when connecting, and something else when not connecting.  So I think all that's required here is to add `STAT_xxx` constants.

peterhinch, 2019-04-09:
That works. Thank you.

dpgeorge, 2019-04-12:
Actually, this won't work fully as expected.  `s.status()` returns 1 when connecting the MAC layer (connecting to the AP) and 2 when it subsequently moves on to connect the IP layer (getting an IP address via DHCP).  So "is-connecting" is defined as `s.status() in (1, 2)` and "is-connected" is defined as `s.status() == 3`.

I'm not sure how to handle this, to make it consistent across the ports.  It could be changed so that states 1 and 2 are merged (to be the same as esp8266 and esp32), but that loses information.  Alternatively, a new function `s.isconnecting()` could be added which has the correct logic for a given port.

peterhinch, 2019-04-12:
As per https://github.com/micropython/micropython/issues/4681, the status remains 1 forever under certain error conditions.