micropython/micropython #4683: Pyboard D: uos.dupterm slows USB REPL character input

peterhinch, 2019-04-08:
Tests used an SF6W with a WBUS-DIP28. Firmware is the 4th April release build and mboot is updated. 
```
MicroPython v1.10-258-g83f3c29d3-dirty on 2019-04-03
```
An FTDI adaptor was connected to UART 1 via the WBUS-DIP28 with a miniterm.py session running. The USB FS interface was connected to a PC running rshell. main.py consisted of:
```python
from machine import UART
import uos
uart = UART(1, 115200)
uos.dupterm(uart)
```
Everything worked as expected, but sometimes character input to rshell slowed to a crawl. Not much faster than 1cps. Characters were buffered: I could type a line but had to wait for the characters to be echoed back and the line executed. This behaviour was not consistent. After a power cycle it would either be in normal or slow mode. Once in a mode, it stayed that way.

dpgeorge, 2019-04-09:
This probably isn't specific to PYBD, since it doesn't really do anything different in this area.

It could be related to the recent `dupterm` changes in 9d6f70f7154aa01a02d3de1c669241e3a1439218

But, my best guess would be simply that the UART needs to be configured in non-blocking mode, like this:
```python
uart = UART(1, 115200, timeout=0)
```
The default timeout is 1000ms, which would correspond to 1cps.  IMO the default should be changed to 0, ie non-blocking by default.

peterhinch, 2019-04-09:
Thank you. I'll try this and see if the problem recurs.

hardkrash, 2019-04-09:
With older code and UART I have seen slowdowns on a STM32 port.
My duped terminal is still fast, the original is slow...

In time Ill add debugging info here or start a new bug, if fixes here don't solve my issue.

peterhinch, 2019-04-10:
So far so good with timeout=0.

hardkrash, 2019-04-18:
The issue is resolved by using the timeout=0 on my case as well.
The timeout being set to 0 on the UART being duplicated to caused the original UART to appear to slow.

pacmac, 2019-04-18:
This issue just started for me with the latest update, anything I key into the REPL appears at a rate of about 1 character per second, CTRL-D does not help.

dpgeorge, 2019-05-07:
The default UART timeout was changed to 0 in 34942d0a72980173eca51b201f271f67bcae46b5

closed 2019-05-07 by dpgeorge