micropython/micropython #4684: PyboardD - Unable to connect to some 2G networks

pacmac, 2019-04-09:
I have an esp8266 and Pyboard-D side by side, the esp8266 connects but the pyboard does not. 

On another 2G network both devices connect successfully.

```
def wltest():
	import time
	from network import WLAN
	WLAN().disconnect()
	WLAN().active(False)
	WLAN().active(True)
	WLAN().connect('2g@mynet','MYPASS')
	while not WLAN().isconnected():
		print('waiting...',WLAN().status())
		time.sleep(1)
	
	print(WLAN().ifconfig())
```

**ESP8266**
```
wltest()
('192.168.42.153', '255.255.255.0', '192.168.42.1', '192.168.42.1')
>>> state: 5 -> 0 (0)
rm 0
pm close 7
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 7
cnt 

connected with xx@xxx, channel 11
dhcp client start...
ip:192.168.42.153,mask:255.255.255.0,gw:192.168.42.1
```

**PYBOARD-D**
```
wltest()
waiting... 1
waiting... -1
waiting... -1
waiting... -1
waiting... -1
waiting... -1
waiting... -1
waiting... -1
waiting... -1
waiting... -1
waiting... -1
waiting... -1
waiting... -1
waiting... -1
waiting... -1
waiting... -1
waiting... -1
waiting... -1
waiting... -1
waiting... -1
```

- Wireless (2.4 GHz / eth1)
- MAC Address	E4:F4:C6:00:6C:3C
- Wireless Mode	Access Point
- Wireless Network Mode	Auto
- Interface Status	Up (LAN)
- Radio	Enabled 
- SSID	wla@xxxxx
- Broadcast	Enabled 
- Security	WPA Personal (PSK) + AES
- Channel	1 - 2.412 GHz
- Channel Width	40 MHz
- Interference Level	Acceptable
- Rate 144mbps

The pyboard continues forever waiting, I have tried setting country to US and GB, wireless is channel width is 40mhz, not sure what other info I can provide ???

dpgeorge, 2019-04-09:
Please enable event logging via `WLAN().config(trace=1)` (do it between active(True) and connect(...)) and report the output.

pacmac, 2019-04-09:
```
wltest()
waiting... 1
[ 7000772] ASYNC(0000,23,0,8,0)
[ 7000772] ASYNC(0000,SET_SSID,1,0,0)
waiting... -1
waiting... -1
waiting... -1
waiting... -1
waiting... -1
waiting... -1
```

pacmac, 2019-04-09:
anything else I can do ?

dpgeorge, 2019-04-09:
```
> [ 7000772] ASYNC(0000,23,0,8,0)
> [ 7000772] ASYNC(0000,SET_SSID,1,0,0)
```
This means the AP was rejected by the STA (the PYBD) because of an RSN mismatch.

Can you please do a `WLAN().scan()` with event tracing enabled, and email me the results?

> Also what are the allowed codes for the pyb.country() and what effect does setting these codes have ?

There are about 235 country codes supported by the chip, and for the most part it restricts the use of certain channels based on the region.  You should be able to use the ones from ISO 3166-1 alpha-2 : https://en.wikipedia.org/wiki/ISO_3166-1_alpha-2 .

pacmac, 2019-04-09:
Email Sent

pacmac, 2019-04-11:
The modified firmware that you sent works.

dpgeorge, 2019-04-11:
The issue here was that it wasn't authenticating correctly with WPA-only (ie non WPA2) access points.  A fix for this is now pushed to the stm32-pybd branch.

closed 2019-04-11 by dpgeorge