micropython/micropython #4687: collections.namedtuple does not support comma in string

jacklinquan, 2019-04-10:
>>> import collections
>>> Point = collections.namedtuple('Point', 'x,y')
>>> Point(1,2)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: function takes 1 positional arguments but 2 were given
>>> Point(1)
Point(x,y=1)
>>> Point = collections.namedtuple('Point', 'x, y') # A space added
>>> Point(1,2)
Point(x,=1, y=2)
>>> # Note a weird parameter name ends with a comma "x,"

jacklinquan, 2019-04-10:
This bug is found in standard firmware for ESP32 port:
esp32-20190125-v1.10.bin

peterhinch, 2019-04-10:
I think the answer to this is that the (u)collections module is a 'micro' implementation with stricter syntax rules than those in the CPython library. Field names must be separated by a space so in your second example the comma is interpreted literally.

dpgeorge, 2019-04-11:
Yes, MicroPython supports only spaces between field names if you pass in a string for the element names.  But it also supports passing in an array or tuple of strings for the elements.

jacklinquan, 2019-04-12:
@peterhinch @dpgeorge 
Thank you all for the replies!

closed 2019-04-12 by jacklinquan