micropython/micropython #4688: error with frozen option

supersjimmie, 2019-04-10:
My issue looks like https://github.com/micropython/micropython/issues/4542 which was only assumed to be solved.

I use an esp32 with:
```~/micropython/ports/esp32$ make FROZEN_MPY_DIR="frozen" ```
The error then is:
```
GEN build/frozen_mpy.c
usage: mpy-tool.py [-h] [-d] [-f] [-q QSTR_HEADER]
                   [-mlongint-impl {none,longlong,mpz}] [-mmpz-dig-size N]
                   files [files ...]
mpy-tool.py: error: the following arguments are required: files
../../py/mkrules.mk:122: recipe for target 'build/frozen_mpy.c' failed
make: *** [build/frozen_mpy.c] Error 2
make: *** Deleting file 'build/frozen_mpy.c'
```
(I don't use symlinks as discussed in the closed issue)

dpgeorge, 2019-04-11:
> Dir ~/micropython/ports/esp32/frozen exists and is empty now.

If the directory is empty then it won't work, there must be at least one .py file in it.  If you don't want any frozen files then use `make FROZEN_MPY_DIR=`

supersjimmie, 2019-04-15:
understood.

closed 2019-04-15 by supersjimmie