micropython/micropython #4690: Pyboard-D Latest Build - ordinal not in range(128)

pacmac, 2019-04-11:
```
make -j6 BOARD=PYBD_SF2
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
make: Warning: File 'mpconfigport.mk' has modification time 0.32 s in the future
GEN build-PYBD_SF2/genhdr/moduledefs.h
../../py/py.mk:346: recipe for target 'build-PYBD_SF2/genhdr/moduledefs.h' failed
Traceback (most recent call last):
  File "../../py/makemoduledefs.py", line 107, in <module>
    main()
  File "../../py/makemoduledefs.py", line 101, in main
    modules |= find_module_registrations(c_file)
  File "../../py/makemoduledefs.py", line 53, in find_module_registrations
    return set(re.findall(pattern, c_file_obj.read()))
  File "/usr/lib/python3.6/encodings/ascii.py", line 26, in decode
    return codecs.ascii_decode(input, self.errors)[0]
UnicodeDecodeError: 'ascii' codec can't decode byte 0xe2 in position 10887: ordinal not in range(128)
make: *** [build-PYBD_SF2/genhdr/moduledefs.h] Error 1
make: *** Deleting file 'build-PYBD_SF2/genhdr/moduledefs.h'
```

pacmac, 2019-04-11:
Also, the main micropython repo fails to build for PBBOARD 1.1.

However ESP8266, ESP32 and NRF all build successfully for me.

closed 2019-04-12 by dpgeorge

dpgeorge, 2019-04-12:
It should now build fine.

pacmac, 2019-04-12:
SF2, SF6, ESP8266, ESP32 and NRF all build successfully

Thanks

PR #4691 (2019-04-12): py/makedefs: Use io.open with utf-8 encoding when processing source.
In case (user) source code contains utf-8 encoded data and the default locale is not utf-8.

Improves upon #4592, and also fixes #4690