micropython/micropython #4692: [NUCLEO L432KC]: UART 1 TX on D5 ?!

mindforger, 2019-04-12:
i tried to use uart and specified 
> from pyb import UART
 u1 = UART(1,115200)
 u1.init(115200, bits=8, parity=0, stop=1)

and exprected to get TX on D10 like given on the reference design, but actually got it on D5

is there a more recent pin map for this specific board or can you tell me how to get infos about the pin mapping?

mindforger, 2019-04-12:
ookay i think i got it now, still a bit confusing :)

https://github.com/micropython/micropython/blob/master/ports/stm32/boards/stm32l432_af.csv

tells ne USART1 PA9, PA10 and B6 and B7

which translates via 
https://github.com/micropython/micropython/blob/master/ports/stm32/boards/NUCLEO_L432KC/pins.csv

to PA9 and PA10 which i could not find  and then D5 and D4

closed 2019-04-12 by mindforger

dhylands, 2019-04-12:
The mpconfigboard.h file is what determines the default pins for a given peripheral.

UART 1 pins are defined here:
https://github.com/micropython/micropython/blob/673e154dfef6aef827b86ea177c211269358b282/ports/stm32/boards/NUCLEO_L432KC/mpconfigboard.h#L28

dhylands, 2019-04-13:
Most peripheral functions are available on multiple pins. The _af.csv file describes all of the available functionality.

The mpconfigboard.h describes the actual mapping used. And the pins.csv file describes the mapping from board pin names to CPU pin names.

So to find out where uart1 pins are, start in mpconfigboard.h which will give you the CPU pin name and then goto pins.csv, which will give the board pin name.