micropython/micropython #4693: [NUCLEO-F767ZI] Timer 1 dead time and break configuration init bug

JuanmaFerrandez, 2019-04-12:
Hi,

I am trying to use PWM with Timer 1 in a Nucleo-F767ZI. But seems dead time and break configuration don't init correctly.

For example:
`tim1 = Timer(1, freq=20000, mode=Timer.UP, deadtime=0, brk=Timer.BRK_OFF)`
`print(tim1)`

Output is:
`Timer(1, freq=20000, prescaler=0, period=10799, mode=UP, div=1, deadtime=848, brk=BRK_HIGH)`

dpgeorge, 2019-04-13:
Thanks for the report, I can confirm the issue.  It looks like the deadtime and break are not being configured properly for the F7 because not all fields of the `TIM_BreakDeadTimeConfigTypeDef` struct are being set.  This then leads to random values being put into the `TIM->BDTR` register which actually locks it from further updates, and gives nonsense values.

dpgeorge, 2019-04-15:
Fixed by 194d6b6788e73216b9c0b60ced0b9ade1b0cb2dd

closed 2019-04-15 by dpgeorge