micropython/micropython #4694: UART readchar writechar

Craigzyc, 2019-04-13:
In the pyboard there is a method on the uart readchar() and writechar() which nicely handle integers.

I am trying to talk to a serial device and can not get it to work with methods in the machine version of uart on other boards.

The device is expecting a packet like
`[66, 1, 32, 1, 3, 0, 0, 11, 0, 1, 4, 102, 198]`
or
`b'\x42\x01\x20\x01\x03\x00\x00\x0B\x00\x04\x02\xE5\x94' `

when I try to write the bytearray to the line as shown it writes 
`b'B\x01 \x01\x03\x00\x00\x0b\x00\x04\x02\xe5\x94'`

In the pyboard version I can do 
`frame = [66, 1, 32, 1, 3, 0, 0, 11, 0, 1, 4, 102, 198]
for ch in frame:
	uart.writechar(ch)
`

Does anyone have a method that would make that work on other boards?

dhylands, 2019-04-13:
Note that b'\x42\x01\x20\x01\x03\x00\x00\x0B\x00\x04\x02\xE5\x94' andb'B\x01 \x01\x03\x00\x00\x0b\x00\x04\x02\xe5\x94' are the same string.

Whenever possible, python (and micropython) will try to use an ASCII character to display a byte string, and will only use hex for characters outside the ASCII range. b'\x42' is the same as b'B' since the letter B has ASCII code 0x42. Similarly b'\x32' and b' ' (single space) are the same.

Craigzyc, 2019-04-13:
Ok, Thank you. I'm using a Saleae logic analyzer to read it and it didn't seem right. I will check again

Craigzyc, 2019-04-13:
Ok, so I hooked up 4 lines using the SALEAE to capture, RX, TX, the TX Enable, and the actual RS485 traffic.
[image]
So the code is like this:
##uart setup
speed = [9600,19200,38400]
receive = 0
send = 1
direction = Pin('P8', mode=Pin.OUT, value=send)
uart = UART(1, speed[2])                         
uart.init(speed[2], bits=8, parity=None, stop=1)

##Send a frame
direction.value(send)
data = [66, 1, 32, 1, 3, 0, 0, 11, 0, 4, 2, 229, 148]
b = bytes(data)
uart.write(b)
direction.value(receive)
##end code
As you can see from the capture the TXenable is not waiting for the transmission to complete. This wasn't an issue on the pyboard because I was sending the frame one character at a time so timing was ok.
If I add time.sleep_ms(10) before the direction.value(receive) the packet finishes sending and the device responds with the appropriate response. I could calculate a time based on packet length to sleep for but it seems hacky.

Note: I tried using the RTS and CTS signals to controls the TXenable on the RS485 transceiver but that doesn't work either.

This is what it looks like with the delay
[image]

###SOLUTION###
For anyone having this issue in the future or for my own personal reference later ;)
The proper way to fix this is to read the tx status of the uart.
Adding this in before the TXenable is removed makes the timing perfect.
`while not uart.wait_tx_done(20):`
`    break`

If it weren't for my logic analyzer I never would have gotten that. Thank you SALEAE

closed 2019-04-13 by Craigzyc