micropython/micropython #4696: Pyboard D - sdio_transfer timeout

pacmac, 2019-04-14:
Thought that I would report that I am starting to receive these errors randomly, only way to overcome is a hard reset:

```
MicroPython v1.9.3-1740-g4fdc731ef-dirty on 2019-04-12; PYBD_SF2W with STM32F722IEK
Type "help()" for more information.
>>> 
>>> 
>>> import io
sdio_transfer timeout STA=0x00000800
sdio_transfer timeout STA=0x00000800
sdio_transfer timeout STA=0x00000800
[CYW43] SDIO enumerate error>>> 
```

```
MicroPython v1.9.3-1740-g4fdc731ef-dirty on 2019-04-12; PYBD_SF2W with STM32F722IEK
Type "help()" for more information.
>>> import io
sdio_transfer timeout STA=0x00400800
sdio_transfer timeout STA=0x00400800
sdio_transfer timeout STA=0x00400800
sdio_transfer timeout STA=0x00400800
sdio_transfer timeout STA=0x00400800
sdio_transfer timeout STA=0x00400800
sdio_transfer timeout STA=0x00400800
sdio_transfer timeout STA=0x00400800
sdio_transfer timeout STA=0x00400800
sdio_transfer timeout STA=0x00400800
sdio_transfer timeout STA=0x00400800
sdio_transfer timeout STA=0x00400800
sdio_transfer timeout STA=0x00400800
...
```

dpgeorge, 2019-04-15:
I assume these messages are seen only after WLAN is initialised, that is after `wl.active(True)` is called?  Are you using both STA and AP interfaces?  How else is the WLAN configured and what is it doing when the errors appear?

pacmac, 2019-04-18:
It's basically the same procedure that I have used for a few years on ESP32/ESP8266

1) Read JSON file from internal flash to load known_nets dictionary.
2) Disable AP
3) Enable STATION
4) Scan Network
5) Perform wlan.config('essid') to see if we already have a ssid
6) if we do have a ssid connect using known_nets dict
7) if we don't then Iterate through scan data and see if any of the SSIDs match known_nets
8) connect to known_net  if found
9) do stuff...
10) deepsleep(5 mins to 1 hour)

But this was intermittent, sometimes it recovers, but mostly it will just repeat the error messages forever.

I am connected via UART and not USB, I have not seen these errors on USB, but that may be a coincidence.

Also, although my procedure wakes from deep sleep, I also saw this error when doing a reset, but I dont recall if that was a hard or soft reset.

What else can I do to help with this ??

dpgeorge, 2019-04-18:
After what step were the errors showing up?

pacmac, 2019-04-18:
too much time has passed, i just don't recall.

I will have to try and replicated it, but could it be related to the recent updates for wakeup etc ?

dpgeorge, 2019-04-18:
Please do try with the very latest code.  If you see the error again please try to take a snapshot of the code used, and firmware, so it can be reproduced.

pacmac, 2019-04-18:
looking back at the original post:

```
>>> import m
sdio_transfer timeout STA=0x00000800
sdio_transfer timeout STA=0x00000800
sdio_transfer timeout STA=0x00000800
[CYW43] SDIO enumerate errorVCC: 3985
io.mode: run
Net-up
```

'm' is my main.py file renamed for debugging, and the outputs of VCC3985 is the reading of the supply voltage to VIN pin, io.mode = run is the result of reading a JSON file (same file that contains the known_nets dict) and Net-up means that wlan is alive.

And in this case, all of these logs happened after the error messages, so it looks like it might be related to the wakeup of some device, possible the wifi chip.

I have seen similar things with the ESP32 and this was a known issue with rev 0 of the chip, and was caused by the flash not being ready when the main code starts up ?!

pacmac, 2019-04-19:
Just happened again:

```
MPY: sync filesystems
MPY: soft reboot
VCC: 3984
io.mode: run
connecting... wlan@symphony A02011962A
Net-up
sdio_transfer timeout STA=0x00400c00
sdio_transfer timeout STA=0x00400c00
sdio_transfer timeout STA=0x00400c00
sdio_transfer timeout STA=0x00400c00
sdio_transfer timeout STA=0x00400c00
sdio_transfer timeout STA=0x00400c00
sdio_transfer timeout STA=0x00400c00
sdio_transfer timeout STA=0x00400c00
sdio_transfer timeout STA=0x00400c00
sdio_transfer timeout STA=0x00400c00
sdio_transfer timeout STA=0x00400c00
sdio_transfer timeout STA=0x00400c00
sdio_transfer timeout STA=0x00400c00
```

Net-up means that isconnected() is True

pacmac, 2019-04-19:
And, after CTRL-D

```
MPY: sync filesystems
MPY: soft reboot
sdio_transfer timeout STA=0x00000800
sdio_transfer timeout STA=0x00000800
sdio_transfer timeout STA=0x00000800
[CYW43] SDIO enumerate errorVCC: 3988
io.mode: run
sdio_transfer timeout STA=0x00000800
sdio_transfer timeout STA=0x00000800
sdio_transfer timeout STA=0x00000800
[CYW43] SDIO enumerate errorconnecting... wlan@symphony A02011962A
sdio_transfer timeout STA=0x00000800
sdio_transfer timeout STA=0x00000800
sdio_transfer timeout STA=0x00000800
[CYW43] SDIO enumerate errorsdio_transfer timeout STA=0x00000800
sdio_transfer timeout STA=0x00000800
sdio_transfer timeout STA=0x00000800
[CYW43] SDIO enumerate errorsdio_transfer timeout STA=0x00000800
sdio_transfer timeout STA=0x00000800
sdio_transfer timeout STA=0x00000800
[CYW43] SDIO enumerate errorsdio_transfer timeout STA=0x00000800
sdio_transfer timeout STA=0x00000800
sdio_transfer timeout STA=0x00000800
[CYW43] SDIO enumerate error[CYW43] send_ethernet failed: -5
sdio_transfer timeout STA=0x00000800
sdio_transfer timeout STA=0x00000800
sdio_transfer timeout STA=0x00000800
[CYW43] SDIO enumerate errorTraceback (most recent call last):
```

pacmac, 2019-04-19:
Is "io" a protected name as I have a module named "io" ??

robert-hh, 2019-04-19:
There is indeed a module call uio, alias io, in MicroPython. It's a subset of CPython's _io.

pacmac, 2019-04-19:
so if I call my module (which contains my gpio pin defs) is this likely to cause problems ?

robert-hh, 2019-04-19:
A short test seems to reveal, that the custom module takes precedence over the builtin. But that may depend on the other factors, e.g. sys.path setting and actual directory when the error occurs.

pacmac, 2019-04-19:
Thanks Robert, yes, I also renamed the module but still face the same issue, so it's not that.

pacmac, 2019-04-21:
I know that it looks like I am just repeating the same errors, bu this is a bit different as I am doing a reset and not loading any other modules:

```
MPY: sync filesystems
MPY: soft reboot
MicroPython v1.9.3-1753-gd1444f538-dirty on 2019-04-18; PYBD_SF2W with STM32F722IEK
Type "help()" for more information.
>>> 
>>> 
>>> 
>>> from network import WLAN
>>> wlan = WLAN(0)
>>> wlan.status()
0
>>> wlan.isconnected()
False
>>> 
>>> wlan.active()
False
>>> wlan.active(1)
sdio_transfer timeout STA=0x00000800
sdio_transfer timeout STA=0x00000800
sdio_transfer timeout STA=0x00000800
[CYW43] SDIO enumerate errorsdio_transfer timeout STA=0x00000800
sdio_transfer timeout STA=0x00000800
sdio_transfer timeout STA=0x00000800
[CYW43] SDIO enumerate errorsdio_transfer timeout STA=0x00000800
sdio_transfer timeout STA=0x00000800
sdio_transfer timeout STA=0x00000800
[CYW43] SDIO enumerate errorsdio_transfer timeout STA=0x00000800
sdio_transfer timeout STA=0x00000800
sdio_transfer timeout STA=0x00000800
[CYW43] SDIO enumerate error[CYW43] send_ethernet failed: -5
>>> 
```

And just to re-iterate I am connected to UART1 and are not using USB.

This is my boot.py:

```
import os
from sys import platform
if platform == 'pyboard':
  import pyb
  if 1==1:
    uart = pyb.UART(1, baudrate=115200)
    os.dupterm(uart)
  pyb.usb_mode('VCP')
  
else:
  from esp import osdebug
  osdebug(None)
```

pacmac, 2019-04-21:
OK, if I switch back to USB no errors, ever.

Switch to UART1 with power to VIN and these errors appear.

pacmac, 2019-04-22:
wow, so many updates !

It now fails while connected via USB, so my last update is not valid.

pacmac, 2019-04-22:
Damien, do you want to login and take a look at this ?

peterhinch, 2019-04-24:
I am seeing this when calling the **publish** method of **umqtt.simple**. The code works as expected apart from emitting these messages each time a publication occurs.
```
sdio_transfer_cmd53: timeout wr=0 len=4 dma=0 buf_idx=0 STA=00000000 SDMMC=00000000:00000000 DMA=00000000:00000000:00000000 RCC=2010017f
sdio_transfer_cmd53: timeout wr=0 len=4 dma=0 buf_idx=0 STA=00000000 SDMMC=00000000:00000000 DMA=00000000:00000000:00000000 RCC=2010017f
[CYW43] error: hdr mismatch a891 ^ 0804
```
Firmware
```
MicroPython v1.10-329-gd1444f538 on 2019-04-18; PYBD_SF2W with STM32F722IEK
```
I get the same results on the SF6W.

dpgeorge, 2019-04-24:
@peterhinch that seems a bit different to what @pacmac experiences, since there the WLAN only recovers with a full power cycle.  I did test an SF2W for 100 hours doing deepsleep(60000) followed by a umqtt publish, in a loop.  It worked fine.  @peterhinch are you able to provide a script which reproduces the error you see?

peterhinch, 2019-04-25:
This demonstrates it. The script is derived from code intended to run from a battery without connection to a PC, hence the LED flashing :)
```python
from pyb import LED, RTC
from umqtt.simple import MQTTClient
import network
import ujson
from  local import SERVER, SSID, PW  # Local configuration: change this file
import uasyncio as asyncio

def publish(s):
    c = MQTTClient('umqtt_client', SERVER)
    c.connect()
    c.publish(b'result', s.encode('UTF8'))
    c.disconnect()

async def wifi_connect(sta_if, red, grn):
    while True:
        red.on()  # Connecting
        grn.off()
        sta_if.active(True)
        sta_if.connect(SSID, PW)
        while sta_if.status() in (1, 2):  # https://github.com/micropython/micropython/issues/4682
            await asyncio.sleep(1)
            grn.toggle()
        if sta_if.isconnected():
            red.off()
            grn.on()
            await asyncio.sleep(1)  # 1s of green == success.
            grn.off() 
            break
        sta_if.deinit()  # Turn off WiFi chip
        await asyncio.sleep(1)  # Try to reconnect

async def main(loop):
    rtc = RTC()
    red = LED(1)
    grn = LED(2)
    count = 0
    sta_if = network.WLAN()
    while True:
        await wifi_connect(sta_if, red, grn)
        publish(ujson.dumps([count, rtc.datetime()]))
        sta_if.deinit()  # Turn off WiFi chip
        count += 1
        await asyncio.sleep(10)

loop = asyncio.get_event_loop()
loop.run_until_complete(main(loop))
```
Firmware
```
MicroPython v1.10-329-gd1444f538 on 2019-04-18; PYBD_SF2W with STM32F722IEK
```

dpgeorge, 2019-04-25:
Thanks @peterhinch , running that script I can see the error you see with the timeout.  Not on every publish but on some of them.

dpgeorge, 2019-04-25:
The bug found above by @peterhinch  should now be fixed by the commit on branch stm32-pybd titled "stm32/cyw43: Update to fix async event handling during deinit.".  The issue was that an asynchronous EXTI event was schedule during the deinit of the WLAN, and then processed after the WLAN was turned off, leading to the error.

peterhinch, 2019-04-25:
Another observation: leaving that script running it has now got stuck at the connecting phase (red LED on, green flashing).

dpgeorge, 2019-04-29:
Thanks to debugging efforts by @pacman the original issue reported here was narrowed down to the use of `machine.freq()` to change system frequency during operation.

The following simple test case will trigger the problem on a PYBD_SF2W:
```python
import machine, network
wl = network.WLAN()
wl.active(1)
for i in range(1000):
    machine.freq(122_000000)
    machine.freq(120_000000)
    print(i, wl.config('mac'))
```

It can also be seen by using just the RNG (on any STM32F722):
```python
import machine, pyb
for i in range(100):
    machine.freq(122_000000)
    machine.freq(120_000000)
    print(i, [pyb.rng() for _ in range(4)])
```
Eventually the RNG will just print 0's (although it'll fix itself on the next loop).  This is because the 48MHz clock source selection is getting stuck and leaving the 48MHz peripherals (RNG, SDMMC, USB) without any clock at all.

The issue is fixed by 8031b7a25c21fb864fe9dd1fa40740030be66c11, with more description in the commit message there.

closed 2019-04-29 by dpgeorge