micropython/micropython #4697: How to use ap + station mode at the same time?

amirmaaref, 2019-04-14:
Hi everyone. 
I need to use esp8266 on Nodemcu in AP+STA mode together.

But when I config AP in my code, I can't connect to another wifi network.
the error is:
can query only one param

What is the problem?
And can I use esp8266 as AP and Station together? is it possible?

peterhinch, 2019-04-14:
I suggest you raise this in [the forum](http://forum.micropython.org/index.php). I think there are people who have made this work.

dpgeorge, 2019-04-15:
Yes it is possible to do this.  Please check the forum and/or docs.

closed 2019-04-15 by dpgeorge

ahmadzein, 2022-08-08:
> Hi everyone. I need to use esp8266 on Nodemcu in AP+STA mode together.
> 
> But when I config AP in my code, I can't connect to another wifi network. the error is: can query only one param
> 
> What is the problem? And can I use esp8266 as AP and Station together? is it possible?

did it work with you?