micropython/micropython #4702: Some strangeness involving range objects with large bounds

ddiminnie, 2019-04-18:
While running some tests on our own fork of CircuitPython we encountered some... interesting... behavior involving range() objects when one (or more) bounds is quite large.  Rather than confuse the issue by citing the behavior of our modified port, the examples below will use the micropython ports we have available to use.

**Example 1**
On the [micropython on-line pyboard-on-unicorn port](http://www.micropython.org/unicorn/) and on win32 builds (windows port, 32-bit release build) range() restricts its arguments to values that fit into a (signed) 32-bit integer:
```
MicroPython cd2f742 on 2017-11-29; unicorn with Cortex-M3
Type "help()" for more information.
>>> range(2**32)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OverflowError: overflow converting long int to machine word
```
However, such range objects can be used in 'for' loops without any exception thrown (or, at least, without an exception being thrown for a LONG time... for my own sanity the example below was interrupted via ctrl-c):
```
MicroPython v1.10 on 2019-04-15; win32 version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> for idx in range(2**32):
...     pass
...
Traceback (most recent call last):
  File "<stdin>", line 2, in <module>
KeyboardInterrupt:
```
An exception may eventually be thrown (hopefully the range index doesn't just overflow to a negative value), but (assuming this is a bug and not an intentional design feature) the bug here is that the interpreter doesn't fail early and loudly.

**Example 2**
On 64-bit unix port (cloned from master 4/18/2019, built on OS X 10.13 using Apple LLVM version 9.0.0 (clang-900.0.39.2)) range objects with large limits behave a bit more strangely:
```
MicroPython v1.10 on 2019-04-18; darwin version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> range(2**32)
range(0, 0)
>>> range(600000000000)
range(0, -1295421440)
```
This behavior is clearly a bug.   'for var in range():' loops with these range objects run for a long time... I'm not sure whether they will ever terminate - certainly any code depending on the loop variable will eventually receive unexpected values.

Oddly enough, the 64-bit Windows port (which still displays 'win32' on startup for some reason...) behaves as expected (though a loop involving such range objects would run a LONG time...):
```
MicroPython v1.10 on 2019-04-18; win32 version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> range(2**32)
range(0, 4294967296)
>>> range(600000000000)
range(0, 600000000000)
```

dpgeorge, 2019-04-19:
> On the pyboard-on-unicorn and on win32 builds range() restricts its arguments to values that fit into a (signed) 32-bit integer
> However, such range objects can be used in 'for' loops without any exception thrown

Yes, that's correct, that's because for loops with immediate values are "optimised" to a while loop with an explicit counting variable.  So it will work correctly.  It will fail with an exception if you don't use an immediate, eg:
```python
x = 2**32
for i in range(x):
    ...
```

> On 64-bit unix port range objects with large limits behave a bit more strangely:

Ok, this is a bug, and nice work for finding it!  But it's just a bug in the printing of range objects (in uPy's custom printf of long ints), not in their actual evaluation.  You can test that it works ok via:
```python
for x in range(1<<60,(1<<60)+10): print(x)
```

> Oddly enough, the 64-bit Windows port behaves as expected

I think that's because, with `_MSC_VER` defined, it uses the system printf and there's no bug with that in the printing.

dpgeorge, 2019-04-19:
See #4703 for a fix

ddiminnie, 2019-04-22:
Thanks for the explanation (it must grow tiresome explaining MicroPython internals to complete strangers... but it is appreciated).

dpgeorge, 2019-04-23:
The printing bug was fixed by 27d22d8712d79636be7115d47c8872d6c1da749c

closed 2019-04-23 by dpgeorge