micropython/micropython #4705: gc_free(...) and gc_realloc(...) could assert failed

fei2020, 2019-04-19:
I embeb upy(HEAD: e0c6dfe90a3e6265b06265c5b3124c24703776f1) into my windows(vs2017 with Windows SDK10.0.17763.0 on windows10) application. 
```
bool __pyDoString(const char* String) {
	nlr_buf_t nlr;
	if (nlr_push(&nlr) == 0) {
		qstr src_name = 1/*MP_QSTR_*/;
		mp_lexer_t *lex = mp_lexer_new_from_str_len(src_name, String, strlen(String), false);
		mp_parse_tree_t pt = mp_parse(lex, MP_PARSE_FILE_INPUT);
		mp_obj_t module_fun = mp_compile(&pt, src_name, MP_EMIT_OPT_NONE, false);
		mp_call_function_0(module_fun);
		nlr_pop();
		return true;
	}
	else {// uncaught exception
		return false;
	}
	return false;
}
```

When I invoke __pyDoString in the main loop(i.e. invoked hundreds of times per second) the assert in gc_free(...) or gc_realloc(...) below will fail. When I modify some macro configs in ports/windows/mpconfigport.h, the assert in gc_realloc(...) does not failed but gc_free(...) will be still. But I don't know which switcher influent it(sorry, I am not familiar with them. I just copy some macros from ports/minimal/mpconfigport.h).
```
assert(ATB_GET_KIND(block) == AT_HEAD); will failed in gc_free(...) and gc_realloc(...)
```
When I set MICROPY_ENABLE_GC to 0, my application could work, but its process memory  keep growing. memory leak when no GC?

The interesting thing is that emscripten version of my application works well with MICROPY_ENABLE_GC (1). the mpconfigport.h of emscripten is based on minimal's.

stinos, 2019-04-20:
This is very likely a duplicate of https://github.com/micropython/micropython/issues/4652

dpgeorge, 2019-04-23:
If link-time optimisation is enabled then try disabling that.

dpgeorge, 2019-05-07:
Should be fixed by 34a7d7ebebc93bf9c4f166b0b523ceab844c7d91

closed 2019-05-07 by dpgeorge