micropython/micropython #4708: network.PPP active = false hangs

nedoskiv, 2019-04-20:
Hello,

I finally managed to connect my GSM modem and start PPP link, but when decide to drop the link in order to control modem again commands hangs (froze the promp) for unlimited time, but can be interrupted with control-C and the link is down after interruption. 

That happen even without actially connect to internet:
```
>>> import machine
>>> import network
>>> gsm = machine.UART(1,tx=22, rx=21, timeout=1000,  baudrate=9600)
>>> ppp=network.PPP(gsm)
>>> ppp.active(True)
True
>>> ppp.active(False)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
KeyboardInterrupt: 
>>> 
>>> 

```

Sometimes it reset module when interrupt with ctrl-c. I'm using:

MicroPython v1.10-290-g8402c26cf on 2019-04-20; ESP32 module with ESP32

dpgeorge, 2019-04-26:
> but when decide to drop the link in order to control modem again prompt/system hangs (froze the promp) for unlimited time,

This would need some debugging, but it could be that it hangs at `xSemaphoreTake(self->inactiveWaitSem, portMAX_DELAY);` which is waiting for the modem to respond to the disconnect command.  If the modem disappeared (or doesn't respond) then this will wait forever.

> I can add that ppp.status() never return anything to me, connected, disconnected does not matter, it always prompt nothing.

This method is not implemented.

> Do not know how to set it UP since there is no documentation.

You need to pass a full 4-tuple, like `ppp.ifconfig(('ip', 'gateway', 'mask', 'dns'))`.  This is documented in http://docs.micropython.org/en/latest/library/network.html#network.AbstractNIC.ifconfig

nedoskiv, 2019-04-26:
Thanks for clearing that.

Hope someone skilled enough pay attention to this issue, because that way,  link is set up one time, and if fail there is no option to reconnect modem since ppp.active(False) hangs the process.

nedoskiv, 2019-05-08:
All right some time was passes and problem is not solved, so here is my proposal (unfortunelly, i'm not skilled enough to fix it myself):
Since problem must be in waiting for modem to respond, let add 2 seconds timeout, if that timeout is reached, function gonna return false, if modem respond inside that 2 seconds function gonna return true
that way programer gonna know if modem is hang and reset it (via reset pin)

dpgeorge, 2019-05-08:
@nedoskiv see #4766 for a potential fix.  I didn't test it with a real PPP device, just a dummy UART.  Please see if it works for you.

nedoskiv, 2019-05-09:
here is my code that usually works to set up PPP link, it is very basic no error checking/etc.:

```
import machine
import time
gsm = machine.UART(1,tx=22, rx=21, timeout=1000,  baudrate=9600)
time.sleep(1)
gsm.write("at\r\n")
print(gsm.readline())
print(gsm.readline())
time.sleep(1)
gsm.write("AT+CPIN?\r\n")
print(gsm.readline())
print(gsm.readline())
time.sleep(1)
#gsm.write("ate0\r\n")
#print(gsm.readline())
#print(gsm.readline())
#time.sleep(1)
gsm.write("AT+CREG?\r\n")
print(gsm.readline())
print(gsm.readline())
time.sleep(1)
gsm.write("AT+CNMI=0,0,0,0,0\r\n")
print(gsm.readline())
print(gsm.readline())
time.sleep(1)
gsm.write('AT+CGDCONT=1,"IP","globul"\r\n')
print(gsm.readline())
print(gsm.readline())
time.sleep(1)
gsm.write('AT+CGDATA="PPP",1\r\n')
time.sleep(1)
print(gsm.readline())
print(gsm.readline())

time.sleep(5)

import network
 
ppp=network.PPP(gsm)
ppp.active(True)
time.sleep(15)

print(ppp.ifconfig())

```
I was begin testing and one time modem fail to respond and system crash:

```
>>> exec(open('./gsm.py').read(),globals())
None
None
None
None
None
None
b'AT+CNMI=0,0,0,0,0\r\r\n'
None
None
None
None
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x40093113  PS      : 0x00060633  A0      : 0x80092495  A1      : 0x3ffe3780  
A2      : 0x0000001d  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000001  
A6      : 0x0000cdcd  A7      : 0x00060023  A8      : 0x00000002  A9      : 0x00000001  
A10     : 0x00000001  A11     : 0x00060623  A12     : 0x00000000  A13     : 0x00000001  
A14     : 0x0000cdcd  A15     : 0x3ffc1460  SAR     : 0x00000008  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000021  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x40093113:0x3ffe3780 0x40092492:0x3ffe37a0 0x40090dcb:0x3ffe37c0 0x4008ed9d:0x3ffe3800 0x4008f1e9:0x3ffe3840 0x400fbcc9:0x3ffe3880 0x400f1340:0x3ffe38d0 0x40189a61:0x3ffe38f0 0x400f21b1:0x3ffe3920 0x40090871:0x3ffe3a50

Rebooting...

```

[python.zip](https://github.com/micropython/micropython/files/3160632/python.zip)

dpgeorge, 2019-05-09:
@nedoskiv not that firmware doesn't contain the fix.  You need to build it yourself using that git branch...

Anyway, you can find the patched firmware here: https://micropython.org/resources/firmware/esp32-ppp-fix.bin

nedoskiv, 2019-05-09:
```
>>> ppp.active()
True
>>> 
>>> exec(open('./mem.py').read(),globals())
GC free mem
115440
>>> ppp.active(False)
False
>>> del ppp
>>> help (ppp)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'ppp' isn't defined
>>> exec(open('./mem.py').read(),globals())
GC free mem
115552
>>> ppp=network.PPP(gsm)
>>> ppp.active(True)
True
>>> Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4011140c  PS      : 0x00060430  A0      : 0x801159dd  A1      : 0x3ffe1130  
A2      : 0x00000000  A3      : 0x00000006  A4      : 0x3ffb5070  A5      : 0x00000000  
A6      : 0x00000004  A7      : 0x00000001  A8      : 0x80113d86  A9      : 0x3ffe1110  
A10     : 0x3ffbb228  A11     : 0x3ffe1110  A12     : 0x3ffb0f9c  A13     : 0x3ffe1100  
A14     : 0x3ffb0f9c  A15     : 0x00000028  SAR     : 0x0000000b  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x000000a2  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x4011140c:0x3ffe1130 0x401159da:0x3ffe1150 0x40115ab9:0x3ffe1170 0x4010b473:0x3ffe11a0 0x40090179:0x3ffe11d0

Rebooting...
ets Jun  8 2016 00:22:57
```
I read that after ppp.active(False) was executed cannot be set true again, so I decided to use it that way, but it crashes, did I do something wrong?

I repeat what I did, and it works next times.

nedoskiv, 2019-05-23:
a question and a suggestion,
is the link to http://micropython.org/resources/firmware/esp32-ppp-fix.bin
rebuild everyday, actually I want to know uart.deinit() gonna be into that image
and suggestion is to be implimented a method for network.ppp deinit.
I hope when script detect that internet is lost, to deinit PPP, uart and start them again, then perhaps everything gonna be fine.

dpgeorge, 2019-05-24:
> is the link to http://micropython.org/resources/firmware/esp32-ppp-fix.bin rebuild everyday

No, it was just a one-off.

> actually I want to know uart.deinit() gonna be into that image

That feature is now part of master so included in the normal daily builds, along with the improvement to closing the PPP connection.

> suggestion is to be implimented a method for network.ppp deinit (or may be ppp.active(false) do it?).

Yes, `ppp.active(False)` is intended to completely deinitialise the link, although it may not work perfectly, and that's pretty much what the issue here is about.

nedoskiv, 2019-05-24:
is now  timeout fix integrated into main?

dpgeorge, 2019-05-24:
> is now timeout fix integrated into main?

Yes, see 0557f0b74b67f91e4ced1c0ffd5d677f1699e64f

MrSurly, 2019-06-12:
With some cellular modules doing `PPP.active(False)` causes the modem to disconnect and return to command mode.  If you wish to restart the connection, then you'll need to perform another dial command, then `PPP.active(True)`

nedoskiv, 2019-06-13:
It is not sure if it can really do that on all modules, so module can be reset (using it reset pin) , and set up for new connection, problem is that micropython crash and restart during the process

ManishSahu53, 2021-07-15:
I am unable to connect my GSM modelu to PPP network. Using TTgo-Tcall 1.4. This is the output.
('0.0.0.0', '0.0.0.0', '255.255.255.255', '0.0.0.0')

Aleksandr66, 2024-12-21:
Maybe the protocol ppp has the concept of Keep Alive?