micropython/micropython #4709: ws2813 not supported in neopixel

toromand, 2019-04-21:
It seems that neopixel supports ws2812b just fine, but does not support the improved ws2813. It seems ws2813 uses different (much longer) RESET timing, and maybe something else.

dpgeorge, 2019-04-26:
Reading some other sources about this (and looking at datasheets) it seems that the only real difference is the longer reset time.  The other bit timings should work on both WS2812 and WS2813.  And a longer reset should also work on the older parts.

So it should be enough to just increase the reset time to get the driver working for old and new.

robert-hh, 2019-04-26:
There is a difference in timing. You can find a timing which suits both devices, but that would be an edge case to one one them. While the total time per bit is the same for both WS2812 and WS2813, the proportions are different, like 1:2 at the WS2812 vs. 1:3 for the WS2813. The longer reset time of the WS2813 is good, as that makes the use of SPI for data transfer more robust.

toromand, 2019-04-26:
The tables above are taken directly from the data sheet(s).
BTW, I believe that the FastLED library actually implemented the compatibility.

robert-hh, 2019-04-26:
Actually I found two versions of the data sheet. The value I cited are form the other one.

dpgeorge, 2019-04-26:
Ok, if there are different timings then it needs a different low-level driver.  What would be good would be to parameterise the existing `neopixel_write()` function so it can take different timing values and support multiple chips.  Other options are to use dedicated hardware to do it, like SPI, but even they require special code to set up the bitstream that goes out on the peripheral.

toromand, 2019-04-26:
Yes, changing the neopixel_write() would be ok, but in order not to break compatibility, and also make it more "user friendly" I would suggest to use an optional enum parameter, something like type=WS2813... and also optional parameters for the actual timings just to be future proof

toromand, 2020-02-29:
Any news on this lately?

big-vl, 2024-01-05:
> Any news on this lately?

So it seems there is a timing parameter, but I haven’t figured out which one to set yet.

jimmo, 2024-01-08:
@big-vl The neopixel driver used on all ports now allows the timings to be specified as a tuple:
https://github.com/micropython/micropython-lib/blob/master/micropython/drivers/led/neopixel/neopixel.py#L17

You can use the values posted by @robert-hh above.

big-vl, 2024-01-16:
> @big-vl The neopixel driver used on all ports now allows the timings to be specified as a tuple: https://github.com/micropython/micropython-lib/blob/master/micropython/drivers/led/neopixel/neopixel.py#L17
> 
> You can use the values posted by @robert-hh above.

```
import machine, neopixel
import time

WS2813 = (300, 1090, 1090, 300)
np = neopixel.NeoPixel(machine.Pin(4), 5, timing=WS2813)
np[0] = (255, 0, 0)
np[1] = (255, 0, 0)
np[2] = (255, 0, 0)
np[3] = (255, 0, 0)
np[4] = (255, 0, 0)
```

Doesn't even include a 5 LED module. What am I doing wrong?

big-vl, 2024-01-16:
```
from machine import Pin, SPI
import gc
import time

class WS2813:
    #buf_bytes = (0x11, 0x13, 0x31, 0x33)
    buf_bytes = (0x88, 0x8e, 0xe8, 0xee)

    def __init__(self, spi_bus, led_count=1, intensity=1, mosi_pin=4):
        self.led_count = led_count
        self.intensity = intensity

        self.buf_length = self.led_count * 3 * 4
        self.buf = bytearray(self.buf_length)
        self.empty_buf = bytearray()

        # SPI init
        self.spi = SPI(spi_bus, baudrate=3200000, polarity=0, phase=1, mosi=Pin(mosi_pin))
        self.spi.init()

        # Turn LEDs off
        self.show(self.empty_buf)

    def show(self, data):
        if data:
            self.fill_buf(data)
        self.send_buf()

    def send_buf(self):
        self.spi.write(self.buf)
        #time.sleep(0.1)
        gc.collect()

    def update_buf(self, data, start=0):
        """
        Fill a part of the buffer with RGB data.
        Order of colors in buffer is changed from RGB to GRB because WS2813 LED
        has GRB order of colors. Each color is represented by 4 bytes in buffer
        (1 byte for each 2 bits).
        Returns the index of the first unfilled LED
        Note: If you find this function ugly, it's because speed optimisations
        beated purity of code.
        """

        buf = self.buf
        buf_bytes = self.buf_bytes
        intensity = self.intensity
        mask = 0x03
        index = start * 12

        for red, green, blue in data:
            red = int(red * intensity)
            green = int(green * intensity)
            blue = int(blue * intensity)

            buf[index] = buf_bytes[green >> 6 & mask]
            buf[index+1] = buf_bytes[green >> 4 & mask]
            buf[index+2] = buf_bytes[green >> 2 & mask]
            buf[index+3] = buf_bytes[green & mask]

            buf[index+4] = buf_bytes[red >> 6 & mask]
            buf[index+5] = buf_bytes[red >> 4 & mask]
            buf[index+6] = buf_bytes[red >> 2 & mask]
            buf[index+7] = buf_bytes[red & mask]

            buf[index+8] = buf_bytes[blue >> 6 & mask]
            buf[index+9] = buf_bytes[blue >> 4 & mask]
            buf[index+10] = buf_bytes[blue >> 2 & mask]
            buf[index+11] = buf_bytes[blue & mask]

            index += 12

        return index // 12

    def fill_buf(self, data):
        end = self.update_buf(data)

        buf = self.buf
        off = self.buf_bytes[0]

        for index in range(end * 12, self.buf_length):
            buf[index] = off

            index += 1

# Создаем экземпляр класса WS2813
led_strip = WS2813(spi_bus=1, led_count=3, intensity=0.1, mosi_pin=4)

# Создаем данные цветов для отображения на ленте
data = [
    (0, 0, 255),
    (255, 255, 255),
    (255, 0, 0),
    #(245, 0, 0),
    #(255, 0, 0),
]
# Отображаем данные на светодиодах
led_strip.show(data)
#time.sleep(1)
#led_strip.show([(0, 85, 0), (0, 0, 255)])
#time.sleep(1)
#led_strip.show([(255, 0, 0), (0, 0, 255)])

```

My test code turns on the module, but there are problems with sequential activation, most likely in sending bytes out of order, they light up only after 1, as if I had 2 LEDs connected in parallel.

big-vl, 2024-02-08:
[image]

The question arises in changing the timings according to this datasheet, well, it turns out to specify the timings from the datasheet, it does not accept it, or the first LED is not lit, does anyone have any manual on how to choose the right timings if I have an oscilloscope, I am a newbie, what do you advise me?

big-vl, 2024-02-11:
I confirm WS2813B V5 works on micropython, all problems are solved, the most important mistake was the incorrect connection circuit. It is impossible for DIN to go to BIN, BIN must be connected to NC or as in the datasheet, BI - BO, why does this happen, the first 24 bytes are not bitten off by the LED, but are transferred to BIN, but BO (NC) is the synchronization channel. This is not just a data transmission channel. Therefore, be careful when connecting LEDs.

I also want to add that there is no need to connect a 470 resistor to the DIN channel, it does not smooth out anything, but only interferes with the connection.

```python
import time
import machine, neopixel
from machine import Pin
from time import sleep_us

high_0 = 280
low_0 = 515
high_1 = 515
low_1 = 745

# Создание кортежа
custom_timing_tuple = (high_0, low_0, high_1, low_1)
pin_led = machine.Pin(14)
np = neopixel.NeoPixel(pin_led, 4, timing=custom_timing_tuple)
np[0] = (0, 0, 0) # set to red, full brightness
np[1] = (0, 0, 0) # set to green, full brightness
np[2] = (0, 0, 0) # set to blue, full brightness
np[3] = (0, 0, 0) # set to red, full brightness

np.write()

time.sleep(0.280)

while True:
    np[0] = (255, 0, 0) # set to red, full brightness
    np[1] = (0, 255, 0) # set to green, full brightness
    np[2] = (0, 0, 255) # set to blue, full brightness
    np[3] = (255, 0, 0) # set to red, full brightness
    np.write()
```

https://github.com/micropython/micropython/assets/60448373/106f64ef-3018-4e43-b7d4-ed9b6a8010e9

[image]

big-vl, 2024-02-11:
> There is a difference in timing. You can find a timing which suits both devices, but that would be an edge case to one one them. While the total time per bit is the same for both WS2812 and WS2813, the proportions are different, like ~1:2 at the WS2812 vs. ~1:3 for the WS2813. The longer reset time of the WS2813 is good, as that makes the use of SPI for data transfer more robust. Edit: The table above for WS2813 seems odd. I found another one which looks more reasonable.
> 
> ```
> T0H    0-code, High-level time      220ns~380ns   -> 300ns +/- 80ns
> T1H    1-code, High-level time      580ns~1.6μs   -> 1090 ns +/- 510ns
> T0L    0-code, Low-level time       580ns~1.6μs   -> 1090 ns +/- 510ns
> T1L    1-code, Low-level time       220ns~420ns   -> 320ns +/- 100 ns
> RES    Frame unit, Low-level time   > 280μs
> ```

There is one important point about synchronizing LEDs, this is the total time, it is not possible to directly specify timings by type, I want it to be 300 here and I want it to be 200 here, there is a total timing time for which no one talks, this is a parameter
Data transfer time: （T0H+T0L、T1H+T1L≥1.25μs)

if you pay attention to my timings, when adding T0H+T0L it will be more than 1.25us, the same applies to T1H+T1L, this is exactly the transmission time. Sorry if this information is of no use to you, but no one has properly explained what even timings actually are.

jonnor, 2024-09-08:
It seems that this has now been implemented and works - as long as one takes care of setting appropriate timings. Thanks a lot for testing and documenting your findings @big-vl !

closed 2024-09-10 by projectgus