micropython/micropython #4713: [RFC] ucryptolib AES-CTR

Jongy, 2019-04-23:
For one of my projects, I've added (a quite poor) support for AES-CTR mode in ``ucryptolib`` (C side). Currently it can be enabled with a ``MICROPY_PY_UCRYPTOLIB_CTR`` config, and when disabled in incurs almost no binary diff.

It doesn't support the `counter` keyword as described in [PEP 272](https://www.python.org/dev/peps/pep-0272/#specification) - only an auto-increment counter with an option to set its initial value.

Is it something you'd like to have? I can tidy it up and make a PR. Or you think this ought to be implemented fully in Python side?

dpgeorge, 2019-04-26:
> I can tidy it up and make a PR. Or you think this ought to be implemented fully in Python side?

If you have the code, it'd be nice to see it to see how much it involves.  Of course you can do everything in Python with just ECB as a primitive, but for performance it is useful to have it in C.

Jongy, 2019-05-07:
Added in ef9843653b795b650e1fe77e22f87e3523bd0a08.

closed 2019-05-07 by Jongy