micropython/micropython #4714: NUCLEO L432KC: pastemode behaves oddly to say the least

mindforger, 2019-04-23:
stemming from that issue here

https://github.com/micropython/micropython/issues/4678

i am trying to load the uPyLoader files manually to the device but in paste mode there is something VERY off, it looks like the inputbuffer is heavily overflowing when pasting longer snippets

i got it working after i pasted it line by line 

but maybe there is a better way to catch this situation or maybe @BetaRavener could add a delay in raw paste mode?!

dpgeorge, 2019-04-26:
There is no flow control on the REPL UART and it's input buffer is just 64 bytes big, so it will certainly have problems with too much data coming in at once.

The input buffer can be increased by defining `MICROPY_HW_UART_REPL_RXBUF` in the mpconfigboard.h file.  Increasing it to 1024 bytes would improve the situation a lot.

Otherwise it's not really possible to fix this because if the host PC sends data faster than the board can keep up then it will inevitably lose data.

mindforger, 2019-04-26:
I will try it out, because upyloader is very handy when one system is behind an ssl breaking firewall

BetaRavener, 2019-04-27:
@dpgeorge Just for info so I can advance with the issue, is this 64 bytes default only related to STM32 port? I haven't heard of people having problems on ESP based boards before.

dpgeorge, 2019-04-28:
>  is this 64 bytes default only related to STM32 port? I haven't heard of people having problems on ESP based boards before.

ESP-based boards have a buffer of 256 bytes and do run into this issue, but usually with larger amounts of input.  So, yes, the 64 bytes is only related to STM32.  Probably it makes sense to increase it from 64 to 256 to improve the pasting behaviour.

mindforger, 2019-04-28:
Sorry, i did not come around today to work on anything

dpgeorge, 2019-05-01:
The UART receive buffer was increased to 260 bytes in cbeac903e86bfb0c50f55960201f9eaa9d45745a, and this should improve the situation with paste.

closed 2019-05-01 by dpgeorge

mindforger, 2019-05-01:
Uhhhh the word alignment would explain the pattern of the failure, didn't know this is a thing with stm