micropython/micropython #4716: H743ZI sleep modes

orelmi, 2019-04-23:
Hello,

I see that function machine_deepsleep has a TODO for STM32H7.

Please could you help me to implement the TODO ? I should do that but i need main guide lines.

Thanks

Aurélien

orelmi, 2019-04-25:
@dpgeorge any idea please ?

dpgeorge, 2019-04-26:
First it's important to understand why `machine.deepsleep()` is not behaving as expected on the STM32H7.  What is it doing wrong?

The TODO in `powerctrl_enter_standby_mode()` is about clearing wake-up flags, which might be accomplished by:
```c
PWR->WKUPCR = 0x3f;
```

iabdalkader, 2019-08-26:
The H7 does not wake up from deepsleep, tried everything even following the example code doesn't work. I'm not sure why, anyway here's the missing flags:

`    PWR->WKUPCR |= PWR_WAKEUP_FLAG1 | PWR_WAKEUP_FLAG2 | PWR_WAKEUP_FLAG3 | PWR_WAKEUP_FLAG4 | PWR_WAKEUP_FLAG5 | PWR_WAKEUP_FLAG6;
`

iabdalkader, 2019-08-27:
It seems that if you have a GPIO configured in EXTI mode deepsleep doesn't work unless you clear the EXTI flag before calling HAL_PWR_EnterSTANDBYMode();
This seems to fix the issue for me:
```
    #elif defined(STM32H7)
    EXTI_D1->PR1 = 0x3fffff;
    PWR->WKUPCR |= PWR_WAKEUP_FLAG1 | PWR_WAKEUP_FLAG2 | PWR_WAKEUP_FLAG3 | PWR_WAKEUP_FLAG4 | PWR_WAKEUP_FLAG5 | PWR_WAKEUP_FLAG6;
    #else
```