micropython/micropython #4718: Eep8266 Setting pin alternate function uart0

qiuwenhui, 2019-04-24:
I want to swap esp8266 UART0 pins (gpio 1 and gpio3) to GPIO13 and GPIO15?How is esp8266 implemented in micropython?Does esp8266 Pin not have Pin.AL mode?

qiuwenhui, 2019-04-24:
uart swap ?Adding both functions enables the user to switch
UART0 Tx from GPIO1 to GPIO15 and Rx from GPIO3
to GPIO13.?
This is another branch, I do not know whether the current version of uart swap similar function?
https://github.com/torwag/micropython/commit/789b96db61b351a50d55409cdfe682f4ea36b704

qiuwenhui, 2019-04-24:
I found this configuration method in BBS and the documentation but why did I get an error
`>>> uart=machine.UART(0,115200,tx=machine.Pin(13),rx=machine.Pin(15)) 
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: extra keyword arguments given`

robert-hh, 2019-04-24:
The keywords rx and tx are disabled in the esp8266 port. It looks like this feature is not implemented.

dpgeorge, 2019-04-26:
It would be possible to implement this feature via the `tx` and `rx` arguments, just check if the given pins are 13/15 and, if so, call `system_uart_swap()`.

dpgeorge, 2019-08-21:
Ability to swap UART pins by specifying tx/rx args was implemented in 96ace8082e409c582911239ee913a95d955058bf

closed 2019-08-21 by dpgeorge