micropython/micropython #4719: [ds18x20 Adresses]: bytearray not proberly assembled

hasenradball, 2019-04-25:
Hi, if I use the the following code to read the ds19x20 sensors:
```
import time, ds18x20
ds = ds18x20.DS18X20(ow)
roms = ds.scan()
ds.convert_temp()
time.sleep_ms(750)
for rom in roms:
    print(ds.read_temp(rom))
```
In roms you will get an array with the bytearrays of the sensors. The array looks like that:

```[bytearray(b' (\xff\xb6\xae\x80\x14\02.')]```

Whats the meaning of the dot at the end?

I think the bytearray is not properly formed from my understanding. Should the bytearray not look like that?

```[bytearray(b'\xff\xb6\xae\x80\x14\02')]```

best regards Frank

dpgeorge, 2019-04-25:
> `[bytearray(b' (\xff\xb6\xae\x80\x14\02.')]`

This looks ok.  (Micro)Python will render bytes as characters if they are printable, so it renders the space, open parenthesis and period.

peterhinch, 2019-04-25:
This is a FAQ in the forum. It is a feature of the Python language which seems to confuse many users.

closed 2019-04-26 by dpgeorge